NeuroTechX/EEG-ExPy #6: select port prompt

basty72, 2020-08-23:
Hi,

IMO, this is a minor bug.
I am following to the available instructions (github/sphynx, or outdated gdoc) and when I come to the serial port declaration, i.e.
```
board_name = 'cyton'
port = '/dev/cu.something'   
eeg_device = EEG(device=board_name,serial_port=port)
```
the latest instruction still prompts a textual instruction for the user to insert the port name again. However, the port name entry is not required and if enter is pressed the code executes correctly.
Just aesthetics, but it'd be great to remove the prompt.

JadinTredup, 2020-08-26:
Can you please link me to the documentation you are talking about?

basty72, 2020-08-26:
sure, pick one:
https://github.com/NeuroTechX/eeg-notebooks

https://neurotechx.github.io/eeg-notebooks/index.html

https://eeg-notebooks.readthedocs.io/en/latest/

https://docs.google.com/document/d/1OHLxsMksb401yjqpwZY2c2YEBA-yn_TGiBASkZ05H3k/edit#heading=h.rxm8dlmqjphd

JadinTredup, 2020-08-26:
The last two of those are dated and should not be being referenced and in the first two I do not see the three lines you mentioned above?

basty72, 2020-08-26:
please see the command line prompt at line 79 of get_openbci_usb():
https://github.com/NeuroTechX/eeg-notebooks/blob/611fa8ca0cadf31b060ffef4a7171831c9c4c06b/eegnb/devices/utils.py#L73

which, for my board (cyton), is called at line 135 of EEG.py
https://github.com/NeuroTechX/eeg-notebooks/blob/master/eegnb/devices/eeg.py

The prompt asks the user to manually insert the usb port name IF serial_port=None (see line 34 of eeg.py), but in my jupyter script I explicitly declare the port string, as in:
`eeg_device = EEG(device=board_name, serial_port=port)`

after which I see a redundant message popping up in my notebook cell:
`Please enter USb port for Mac OS`

Then I press enter and all works well, however this might be confusing

JohnGriffiths, 2020-08-27:
The message is redundant because the board specific serial port definitions such as for `cyton` [here](https://github.com/NeuroTechX/eeg-notebooks/blob/611fa8ca0cadf31b060ffef4a7171831c9c4c06b/eegnb/devices/eeg.py#L135), which in your case asks for a port specification if OS is `"Darwin"`, is overwritten slightly later [here](https://github.com/NeuroTechX/eeg-notebooks/blob/611fa8ca0cadf31b060ffef4a7171831c9c4c06b/eegnb/devices/eeg.py#L163) by the port number argument provided (if it is provided) when the class is initialized.

This is indeed redundant and also a bit of a red herring in terms of code readability. 

Fixing now...

JohnGriffiths, 2020-08-27:
Added a fix - https://github.com/NeuroTechX/eeg-notebooks/commit/06fc10ad869b4b23492e82547aed789a990e897c

@basty72  Has the redundant prompt stopped now?

JadinTredup, 2020-09-06:
@JohnGriffiths I am pretty certain we can mark this as closed now?

basty72, 2020-09-08:
closed for me

closed 2020-09-11 by basty72