NeuroTechX/EEG-ExPy #7: error downloading cueing data during doc build

kylemath, 2020-08-25:
Traceback (most recent call last):
  File "/Users/kylemathewson/eeg-notebooks/venv/lib/python3.7/site-packages/sphinx_gallery/gen_gallery.py", line 159, in call_memory
    return 0., func()
  File "/Users/kylemathewson/eeg-notebooks/venv/lib/python3.7/site-packages/sphinx_gallery/gen_rst.py", line 466, in __call__
    exec(self.code, self.fake_main.__dict__)
  File "/Users/kylemathewson/eeg-notebooks/examples/visual_cueing/02r__cueing_group_analysis.py", line 38, in <module>
    datasets.fetch_dataset(data_dir=eegnb_data_path, experiment='visual-cueing', site='kylemathlab_dev')
  File "/Users/kylemathewson/eeg-notebooks/eegnb/datasets/datasets.py", line 85, in fetch_dataset
    with zipfile.ZipFile(destination, 'r') as zip_ref:
  File "/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/zipfile.py", line 1258, in __init__
    self._RealGetContents()
  File "/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/zipfile.py", line 1325, in _RealGetContents
    raise BadZipFile("File is not a zip file")
zipfile.BadZipFile: File is not a zip file

JohnGriffiths, 2020-08-25:
Let's try to debug this dataset fetcher thing outside of the doc build. 

1. Retry on command line 

```python
datasets.fetch_dataset(data_dir=sometestfolderpath, experiment='visual-cueing', site='kylemathlab_dev')
```

2. Is the dataset fetcher working ok for the other datasets?

JadinTredup, 2020-08-26:
Is this okay to close with the PR merge?

JohnGriffiths, 2020-08-27:
I'm also getting this error now - seems the new dataset fetcher is not working with the `cueing` experiment dataset. Probably because of the zip file size.

JohnGriffiths, 2020-08-27:
Changed the dataset fetcher to use the `gdown` package, - https://github.com/NeuroTechX/eeg-notebooks/commit/f11fa0c86aead8122225295846d6f141cd707f22

this seems to have no problem with large file downloads. It's also about 1% the number of lines :) 
...but an additional dependency :(

@kylemath - try the dataset fetcher commands above now.

JadinTredup, 2020-08-27:
Thats interesting, I tried it earlier and it worked fine.

kylemath, 2020-08-27:
one liner above didn't run, must need to import some things first, so I tried to remake the html docs

I pulled your change from upstream repo, merged into my branch, pip install gdown, I also added gdown to the requirements-docs.txt, then try to make the htmls and get:

Traceback (most recent call last):
  File "/Users/kylemathewson/eeg-notebooks/venv/lib/python3.7/site-packages/sphinx_gallery/gen_gallery.py", line 159, in call_memory
    return 0., func()
  File "/Users/kylemathewson/eeg-notebooks/venv/lib/python3.7/site-packages/sphinx_gallery/gen_rst.py", line 466, in __call__
    exec(self.code, self.fake_main.__dict__)
  File "/Users/kylemathewson/eeg-notebooks/examples/visual_cueing/01r__cueing_singlesub_analysis.py", line 30, in <module>
    from eegnb.datasets import fetch_dataset
  File "/Users/kylemathewson/eeg-notebooks/eegnb/datasets/__init__.py", line 2, in <module>
    from .datasets import fetch_dataset
  File "/Users/kylemathewson/eeg-notebooks/eegnb/datasets/datasets.py", line 5, in <module>
    import gdown
ModuleNotFoundError: No module named 'gdown'

JohnGriffiths, 2020-08-27:
So what do you get on the command line with just 

```bash
ipython
```
```python
import gdown
```

?

kylemath, 2020-08-27:
> So what do you get on the command line with just
>
> ipython
>
> import gdown
>
> ?

import gdown works when I :

cd eeg-notebooks
source venv/bin/activate
python
>>>import gdown

-- 
*Kyle E. Mathewson, Ph.D.*
Assistant Professor - Department of Psychology, Faculty of Science
Director - Attention Perception and Performance Lab
Affiliate - Neuroscience and Mental Health Institute, Faculty of Medicine
and Dentistry
University of Alberta
P455 - Biological Sciences Building
11455 Saskatchewan Dr.
Edmonton, Alberta, Canada, T6G 2E9
Phone: 1-780-492-2662
Email: kyle.mathewson@ualberta.ca
Web: www.kylemathewson.com

JohnGriffiths, 2020-08-27:
So was it looking for a different python installation when you were building before?

Does this work now?
```bash
source venv/bin/activate
cd doc
make html
```

kylemath, 2020-08-27:
yeah it works, but that is what produces the error in the html that gdown
was not available

-- 
*Kyle E. Mathewson, Ph.D.*
Assistant Professor - Department of Psychology, Faculty of Science
Director - Attention Perception and Performance Lab
Affiliate - Neuroscience and Mental Health Institute, Faculty of Medicine
and Dentistry
University of Alberta
P455 - Biological Sciences Building
11455 Saskatchewan Dr.
Edmonton, Alberta, Canada, T6G 2E9
Phone: 1-780-492-2662
Email: kyle.mathewson@ualberta.ca
Web: www.kylemathewson.com

JohnGriffiths, 2020-08-27:
Must be trying to use a different python installation then, if the import isn't working.

JohnGriffiths, 2020-11-25:
@kylemath - checking up on this. We have tried to iron out the dependencies in `requirements.txt` a bit more.  

please try: 

1. fresh git clone, or pull updates

2. create fresh conda or venv 

3. pip install

`pip install -e .`

4. build docs as above

`make html`

Are you still getting an error?

kylemath, 2020-11-26:
```Successfully installed eeg-notebooks
(venv) Kyles-MacBook-Pro:eeg-notebooks kylemathewson$ make html
make: *** No rule to make target `html'.  Stop.
(venv) Kyles-MacBook-Pro:eeg-notebooks kylemathewson$ cd docs
-bash: cd: docs: No such file or directory
(venv) Kyles-MacBook-Pro:eeg-notebooks kylemathewson$ ls
LICENSE			eegnb			requirements_test.txt
README.rst		examples		run_notebooks.py
binder			requirements-docs.txt	setup.cfg
doc			requirements-noeeg.txt	setup.py
eeg_notebooks.egg-info	requirements.txt	venv
(venv) Kyles-MacBook-Pro:eeg-notebooks kylemathewson$ cd doc
(venv) Kyles-MacBook-Pro:doc kylemathewson$ ls
Makefile	changelog.rst	docs_notes.md	getting_started	index.rst
_static		conf.py		experiments	img		misc
(venv) Kyles-MacBook-Pro:doc kylemathewson$ make html
/bin/sh: sphinx-build: command not found
make: *** [html] Error 127
(venv) Kyles-MacBook-Pro:doc kylemathewson$ 
```

JohnGriffiths, 2020-11-29:
Apols.

Ok I've added the doc building dependencies into the `requirements.txt`. That's easier than maintaining multiple installation instructions. 

Pls `git pull` master and try `pip install -e .` again. 

Note that the env needs to be with Python 3.6.

tmorshed, 2024-04-04:
closed as is stale - Python versions and installations have changed ever since and `gdown` issues are now solved, until `gdown` pushes another breaking update!

closed 2024-04-04 by tmorshed