NeuroTechX/EEG-ExPy #9: cueing notebook prose updates

kylemath, 2020-08-25:
There's a set of sphinx-gallery examples for the four notebooks that you contributed

https://neurotechx.github.io/eeg-notebooks/experiments/cueing.html  

which are slightly modified, but largely unchanged from your original notebooks. 

What we could do with is a go-over of those four pages

https://neurotechx.github.io/eeg-notebooks/auto_examples/visual_cueing/01r__cueing_singlesub_analysis.html# 

https://neurotechx.github.io/eeg-notebooks/auto_examples/visual_cueing/02r__cueing_group_analysis.html 

https://neurotechx.github.io/eeg-notebooks/auto_examples/visual_cueing/03r__cueing_behaviour_analysis_winter2019.html

https://neurotechx.github.io/eeg-notebooks/auto_examples/visual_cueing/04r__cueing_group_analysis_winter2019.html

 
Specifically: I was wondering if someone from your team or yourself could spare maybe an hour or so going through those and get back to me with the following:

- Some additional explanatory text for the experiment(s) in general
- Explanatory text for each code block section 
- Explanatory text describing what the figures are showing
- A short concluding paragraph for each page
- Citations of relevant papers (yours and others) in the above additions highly encouraged

If you guys don't have bandwidth for this atm, no worries. I thought I would ask as you folks would be best placed to do this, and I have a lot of other tasks to do with the rest of the documentation. Trying to get better at delegating!

If you think you could give this a stab: 

You can send the above info however is most convenient. 

Pull request on the source code would be ok - although the relationship of sphinx-gallery python source code embedded .rst to the webpage html isn't super obvious, and you have to compile the website to see it. So you might prefer to e.g. just grab the jupyter notebooks from  each webpage (c.f. download link at the bottom), and email me updated versions, or even just text file/email. Whatever you prefer.

kylemath, 2020-08-25:
working on this in https://github.com/NeuroTechX/eeg-notebooks/pull/8

JadinTredup, 2020-08-26:
Is this okay to close with the PR merge?

kylemath, 2020-08-26:
Still working on it

kylemath, 2020-08-27:
cd doc
<make your changes>
make html

cd _build/html
python -m http.server 8002

in web browser:
local:8002

JohnGriffiths, 2020-08-27:
yep.

JohnGriffiths, 2020-09-11:
@kylemath any progress on this?

kylemath, 2020-09-11:
hmm, no sorry, where was I, I have added to the main page, but need to edit each of the individual ones, and I couldn't get the docs to compile
also one of the four may be redundant, 

I think they are 1) single subject analysis 2) group analysis 3) group analysis behaviour 4) group analysis compare two semesters (this one is probably the most expendable)

JohnGriffiths, 2020-10-04:
Are the compilation errors due to .rst errors?

The syntax for defining text/code sections with sphinx can be a bit fiddly. 

Suggest either a) start back with the original code and add changes incrementally and check it doesn't break along the way, and/or b) send a description of the changes you would like to make and we can get implemented and working

kylemath, 2020-10-05:
things are looking pretty good on the master now, what more do you want
prose for, the individual 3 modules?

-- 
*Kyle E. Mathewson, Ph.D.*
Associate Professor - Department of Psychology, Faculty of Science
Director - Attention Perception and Performance Lab
Affiliate - Neuroscience and Mental Health Institute, Faculty of Medicine
and Dentistry
University of Alberta
P455 - Biological Sciences Building
11455 Saskatchewan Dr.
Edmonton, Alberta, Canada, T6G 2E9
Phone: 1-780-492-2662
Email: kyle.mathewson@ualberta.ca
Web: www.kylemathewson.com

JohnGriffiths, 2020-11-25:
Apols for delay, coming back to this. 

Hopefully the docs compilation is working ok now. 

Re: the content - 

See the first comment at the top of this thread. 

What we would like is to have for each of the 4 cueing examples

https://neurotechx.github.io/eeg-notebooks/auto_examples/visual_cueing/01r__cueing_singlesub_analysis.html

https://neurotechx.github.io/eeg-notebooks/auto_examples/visual_cueing/02r__cueing_group_analysis.html

https://neurotechx.github.io/eeg-notebooks/auto_examples/visual_cueing/03r__cueing_behaviour_analysis_winter2019.html

https://neurotechx.github.io/eeg-notebooks/auto_examples/visual_cueing/04r__cueing_group_analysis_winter2019.html

is 

1. At the top of the page, a para describing the experiment and analyses to be done, and what they're for

2. After each figure, a description + explanation of what it shows

3. At the bottom of each page, a wrap-up para summarizing what we done and how it was shown and what it means

Would that be possible?

tmorshed, 2024-04-04:
closing as merged (#8).

closed 2024-04-04 by tmorshed