NeuroTechX/EEG-ExPy #12: finalize requirements.txt and pip install

JohnGriffiths, 2020-08-29:
The top priority remaining from the version 0.2 refactor is a clear, fully tested, and fully documented installation process. 

The focus point for this is the [requirements.txt](https://github.com/NeuroTechX/eeg-notebooks/blob/master/requirements.txt). 

---

Side note: the developer installation method should be 

`pip install -e . `

from the repo top level folder. 

This command calls the `setup.py` file, which [reads dependencies from the requirements file](https://github.com/NeuroTechX/eeg-notebooks/blob/master/setup.py#L8)

---

The `requirements.txt` needs to be extended to include the following:

- [ ] Version numbers for all packages (e.g. `pip install matplotlib==3.0.3`)
- [ ] Operating system dependencies 
(e.g. `pyreadline==2.1; platform_system == "Windows"` - see [here](https://stackoverflow.com/questions/16011379/operating-system-specific-requirements-with-pip) and [here](https://www.python.org/dev/peps/pep-0508/#environment-markers)
- [ ] Testing of of all this across all operating systems and multiple devices
- [ ] Removal of any redundancies

Finalizing this is number one top priority.

JohnGriffiths, 2020-08-29:
@mhough and @JadinTredup - would be great if you two could take the lead on this, as you are (I believe) the ones with access to all OS types.

JadinTredup, 2020-09-03:
@JohnGriffiths I will try on Windows and Linux but I do not have a MacOS.

JadinTredup, 2020-09-06:
Just wanted to update this for the sake of documenting it: the current working set of requirements for both Windows and Linux (with version numbers) is in "requirements_test.txt" once it is confirmed to work on Mac as well we can move it to the main requirements file.

jartuso, 2020-09-10:
I'm installing on Windows in a new Anaconda environment ((conda 4.8.4) using:

`pip install -r requirements.txt`

I get the following error during the installation:

`ERROR: psychopy 2020.2.4.post1 has requirement pyglet>=1.3, but you'll have pyglet 1.2.0 which is incompatible.`

but the installation continues after that error it looks like psychopy ends up getting installed successfully: 

`Running setup.py install for psychopy ... done`
`Successfully installed...psychopy-2020.2.4.post1...pyglet-1.2.0...`

full log of installation:
[conda-log.txt](https://github.com/NeuroTechX/eeg-notebooks/files/5203452/conda-log.txt)

jartuso, 2020-09-10:
Following up -- I had issues running N170 experiment after installing using requirements.txt 

The N170 session would start, but would only display a blank gray screen. None of the faces or houses would show up.
I uninstalled pyglet 1.2.0 (pip uninstall pyglet) and then re-installed using pip install pyglet
Pyglet 1.5.7 was installed. Then the N170 experiment ran normally. 

I made a new environment and modified the requirements.txt to not include a specific pyglet version. The pip installation then ran without errors, and the N170 worked first try in the new environment

JadinTredup, 2020-09-10:
@jartuso sorry I should have mentioned somewhere that we should be trying out "requirements_test.txt" as the current list of requirements. I will move them over to the main requirements file.

JadinTredup, 2020-09-10:
Do we have anyone who has verified if these requirements work on Mac yet?

m9h, 2020-09-11:
I can confirm a clean pip install on Mac OS X 10.14.6 with a fresh miniconda 3.8 using requirement_test.tx. On Mac OS X, conda install python and pip or it will use the /usr/bin/python following current directions.

Runs n170 experiment successfully with port set to /dev/cu.usbserial-DB00M80D 

raw = load_data(save_fn) still doesn't work with positional argument errors. Will follow up. 

Tested on both MacBook Air with USB port and MacBookPro with USB-C USB Hub

closed 2020-09-11 by JadinTredup