NeuroTechX/EEG-ExPy #14: Clearing old examples

JadinTredup, 2020-09-11:
We should go through the old examples and delete the ones that reference the old API. 

@JohnGriffiths are there any reasons we need to keep any of these examples?

JohnGriffiths, 2020-10-04:
No, they can go. Or, better, be updated.

The place for scratch / temp / possibly to be removed things is the sandbox folder.

JohnGriffiths, 2020-12-02:
resuming and executing this now. will also put together a downloader for the compiled notebooks from the website.

JohnGriffiths, 2020-12-02:
done. 

summary: 

- removed .ipynb files in visual n170, visual p300, and visual ssvep folders

- the place for old .ipynbs is the sandbox folder

- however there are still some .ipynbs in the visual cueing folder. I'm not sure we want to remove those yet. 

l

closed 2020-12-02 by JohnGriffiths