NeuroTechX/EEG-ExPy #15: OpenBCI Stream Not Detected on Windows

AyrusCode, 2020-09-14:
I first started the OpenBCI livestream using the GUI via dongle. I then go to eeg-notebooks and run the following cell:
```
board_name = 'cyton_daisy'
experiment = 'auditory_oddball'
subject_id = 1
session_nb = 1
#record_duration=1000
record_duration=30

eeg_device = EEG(device=board_name)

# Create save file name
save_fn = generate_save_fn(board_name, experiment, subject_id, session_nb)
print(save_fn)
```
I am then given a prompt for the USB port for Windows and after entering `COM3` I get the following error:
```
---------------------------------------------------------------------------
BrainFlowError                            Traceback (most recent call last)
<ipython-input-18-591d2bca8dd7> in <module>
      6 record_duration=30
      7 
----> 8 eeg_device = EEG(device=board_name)
      9 
     10 # Create save file name

c:\users\surya\documents\eegnote\eegnotesep2020\eeg-notebooks-master\eegnb\devices\eeg.py in __init__(self, device, serial_port, serial_num, mac_addr, other, ip_addr)
     48         self.other = other
     49         self.backend = self._get_backend(self.device_name)
---> 50         self.initialize_backend()
     51 
     52     def initialize_backend(self):

c:\users\surya\documents\eegnote\eegnotesep2020\eeg-notebooks-master\eegnb\devices\eeg.py in initialize_backend(self)
     52     def initialize_backend(self):
     53         if self.backend == 'brainflow':
---> 54             self._init_brainflow()
     55         elif self.backend == 'muselsl':
     56             self._init_muselsl()

c:\users\surya\documents\eegnote\eegnotesep2020\eeg-notebooks-master\eegnb\devices\eeg.py in _init_brainflow(self)
    181         self.sfreq = BoardShim.get_sampling_rate(self.brainflow_id)
    182         self.board = BoardShim(self.brainflow_id, self.brainflow_params)
--> 183         self.board.prepare_session()
    184 
    185     def _start_brainflow(self):

c:\users\surya\anaconda3\envs\eegnbsep2020\lib\site-packages\brainflow\board_shim.py in prepare_session(self)
    808         res = BoardControllerDLL.get_instance ().prepare_session (self.board_id, self.input_json)
    809         if res != BrainflowExitCodes.STATUS_OK.value:
--> 810             raise BrainFlowError ('unable to prepare streaming session', res)
    811 
    812     def start_stream (self, num_samples: int = 1800 * 250, streamer_params: str = None) -> None:

BrainFlowError: UNABLE_TO_OPEN_PORT_ERROR:2 unable to prepare streaming session

```

JadinTredup, 2020-10-02:
Hi @AyrusCode sorry for the late reply, this didn't go to my email for some reason. To answer your question, you should not have the live stream running in the OpenBCI GUI while running EEG notebooks. The point of opening the GUi is just to see what USB port the dongle is being read from; it changes from system to system. Once you know its COM3 then you can close the GUI and run the notebooks.

JadinTredup, 2021-02-13:
@AyrusCode have you re-tried this/can I mark as closed?

closed 2021-02-22 by JadinTredup