NeuroTechX/EEG-ExPy #16: EEG experiments not working with Muse2 via EEG Notebooks on Mac

kushaalrao, 2020-10-03:
I have created a virtual environment and I am trying to run the SSVEP experiment via EEG notebooks. On my command line, I execute the command "python run_notebook.py" and choose the appropriate device and configurations. It connects to the Muse and the stimulus presentation appears, but it doesn't seem to record and store any of the data and I also get an error at the end "Value error: need at least one array to concatenate". 
[image]

[image]

JohnGriffiths, 2020-10-04:
Hi Kushal. 

One possibility here is that the experiment setup processes are taking longer than 10s to start, and so the record process is timing out before any data is saved. 

Could you repeat with, say, 30s for the duration option. Same error?

JohnGriffiths, 2020-10-04:
Also, do you get the same error for all 3 experiment options?

JadinTredup, 2020-10-05:
Interestingly enough I am getting the exact opposite error now. The data is recording but no stimulus is shown when using the Muse2.

JadinTredup, 2020-10-05:
@kushaalrao I just pushed a fix that remedied my errors. Could you update and rerun and see if you errors still persist? If so, I can begin to figure out what is wrong there.

kushaalrao, 2020-10-21:
[image]
I ran it again with a recording duration of 30 seconds, and it says that it was able to successfully run and record the file. However, I can't seem to access the file path that it stored the recording in. When I try to step  through the directory in terminal. it says no such file or directory exists

andraderenew, 2021-05-18:
I have the same issue sort of. I get **no file recorded** although the error it outputs is different. And this depends on the mode of execution. Let's be more precise. If I run in terminal 'eegnb runexp -ip' then I input all the values correctly and  I get the PTB sound from psychopy error. 

EEG device successfully connected!
Press [ENTER] when ready to begin...
Traceback (most recent call last):
  File "/Users/andraderenew/opt/anaconda3/envs/eeg-notebooks/bin/eegnb", line 33, in <module>
    sys.exit(load_entry_point('eeg-notebooks', 'console_scripts', 'eegnb')())
  File "/Users/andraderenew/opt/anaconda3/envs/eeg-notebooks/lib/python3.7/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/Users/andraderenew/opt/anaconda3/envs/eeg-notebooks/lib/python3.7/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/Users/andraderenew/opt/anaconda3/envs/eeg-notebooks/lib/python3.7/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/andraderenew/opt/anaconda3/envs/eeg-notebooks/lib/python3.7/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/andraderenew/opt/anaconda3/envs/eeg-notebooks/lib/python3.7/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/Users/andraderenew/eeg-notebooks/eegnb/cli/__main__.py", line 55, in runexp
    run_introprompt()
  File "/Users/andraderenew/eeg-notebooks/eegnb/cli/introprompt.py", line 132, in main
    run_experiment(experiment, record_duration, eeg_device, save_fn)
  File "/Users/andraderenew/eeg-notebooks/eegnb/cli/utils.py", line 38, in run_experiment
    module.present(duration=record_duration, eeg=eeg_device, save_fn=save_fn)  # type: ignore
  File "/Users/andraderenew/eeg-notebooks/eegnb/experiments/visual_ssvep/ssvep.py", line 20, in present
    record_duration = np.float32(duration)
TypeError: float() argument must be a string or a number, not 'EEG'
0.7603 	WARNING 	PTB audio lib was requested but not loaded: psychtoolbox audio failed to import
0.7863 	WARNING 	For experiments that use audio stimuli, timing will be much better if you upgrade your PsychoPy installation to a 64bit Python3 installation and use the PTB backend.

Also I have in this issue one doubt. **Muse is never connected correctly** although it says here in my code that it connected correctly. 

Also I can point out that if I execute the code from python console then the experiments run and it finishes but it never saves and data recorded. My guess is because it never actually gets connected properly. I do have all libraries required. I even have python 3.7 although I run everything in **ARM processor M chip apple big sur 11.3.1**.

closed 2021-05-20 by oreHGA

andraderenew, 2021-05-20:
I don't quite follow! Can anyone explain how will this be solved? Sorry for not knowing much of the dynamics here on github.

oreHGA, 2021-05-20:
No worries @andraderenew .  Can you try the following:

- Pull/Download the latest version of the eeg-notebooks project again like you did when you first started (this will give you the latest changes that resolves your issue)

- try running the experiment again.

andraderenew, 2021-05-20:
ah ok thank you!! I guessed as there were a few files changed 22mins ago that possibly that could be it but I wanted to make sure

PR #101 (2021-05-20): fix: fixed incorrect parameter order
Fixes #16