NeuroTechX/EEG-ExPy #20: issue with installation

danielemarinazzo, 2020-11-05:
Hi
I am trying to install the library on windows, starting from scratch with miniconda (python 3.9).

I get a series of errors, I paste the first few lines below, I think it might be a version issue.

Any hint?

thanks!

  ERROR: Command errored out with exit status 1:
   command: 'C:\Users\ricercasc\Miniconda3\envs\eeg-notebooks\python.exe' 'C:\Users\ricercasc\Miniconda3\envs\eeg-notebooks\lib\site-packages\pip' install --ignore-installed --no-user --prefix 'C:\Users\ricercasc\AppData\Local\Temp\pip-build-env-8aype7ym\overlay' --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- setuptools wheel 'Cython>=0.28.5' 'numpy==1.13.3; python_version=='"'"'3.6'"'"' and platform_system!='"'"'AIX'"'"' and platform_python_implementation == '"'"'CPython'"'"'' 'numpy==1.14.0; python_version=='"'"'3.6'"'"' and platform_system!='"'"'AIX'"'"' and platform_python_implementation != '"'"'CPython'"'"'' 'numpy==1.14.5; python_version=='"'"'3.7'"'"' and platform_system!='"'"'AIX'"'"'' 'numpy==1.17.3; python_version>='"'"'3.8'"'"' and platform_system!='"'"'AIX'"'"'' 'numpy==1.16.0; python_version=='"'"'3.6'"'"' and platform_system=='"'"'AIX'"'"'' 'numpy==1.16.0; python_version=='"'"'3.7'"'"' and platform_system=='"'"'AIX'"'"'' 'numpy==1.17.3; python_version>='"'"'3.8'"'"' and platform_system=='"'"'AIX'"'"'' 'scipy>=0.19.1'
       cwd: None
  Complete output (641 lines):
  Ignoring numpy: markers 'python_version == "3.6" and platform_system != "AIX" and platform_python_implementation == "CPython"' don't match your environment

JohnGriffiths, 2020-11-05:
Hi Daniele. 

Great to hear from you! Nice to hear you are interested in eeg-notebooks. 

The installation was mainlyl tested with python 3.7 and python 3.8 so far. 

Could you please try with 3.8?

`conda create -n "eeg-notebooks_py38" python=3.8`

`conda activate eeg-notebooks_py38`

`conda install git`

`git clone https://github.com/neurotechx/eeg-notebooks`

`cd eeg-notebooks`

`pip install -e .`

(Note that this last step is different to the docs, which currently say `pip install -r requirements.txt` . This needs updating. The above is correct )

( p.s. we will be doing an eeg-notebooks sprint for brainhack global in early Dec. Let us know if you'd like to join. More info coming soon.  )

danielemarinazzo, 2020-11-06:
thanks a lot, it worked!!

closed 2020-11-06 by danielemarinazzo