NeuroTechX/EEG-ExPy #21: pylsl missing library?

danielemarinazzo, 2020-11-06:
Hi again, another (newbie) issue, with a missing library.
Here the output when running python run_notebooks.py

Traceback (most recent call last):
  File "run_notebooks.py", line 4, in <module>
    from eegnb.devices.eeg import EEG
  File "C:\Users\ricercasc\eeg-notebooks\eegnb\devices\eeg.py", line 18, in <module>
    from muselsl import stream, list_muses, record
  File "C:\Users\ricercasc\Miniconda3\envs\eeg-notebooks_py38\lib\site-packages\muselsl\__init__.py", line 1, in <module>
    from .stream import stream, list_muses
  File "C:\Users\ricercasc\Miniconda3\envs\eeg-notebooks_py38\lib\site-packages\muselsl\stream.py", line 2, in <module>
    from pylsl import StreamInfo, StreamOutlet
  File "C:\Users\ricercasc\Miniconda3\envs\eeg-notebooks_py38\lib\site-packages\pylsl\__init__.py", line 2, in <module>
    from .pylsl import IRREGULAR_RATE, DEDUCED_TIMESTAMP, FOREVER, cf_float32,\
  File "C:\Users\ricercasc\Miniconda3\envs\eeg-notebooks_py38\lib\site-packages\pylsl\pylsl.py", line 1217, in <module>
    lib = CDLL(libpath)
  File "C:\Users\ricercasc\Miniconda3\envs\eeg-notebooks_py38\lib\ctypes\__init__.py", line 381, in __init__
    self._handle = _dlopen(self._name, mode)
FileNotFoundError: Could not find module 'C:\Users\ricercasc\Miniconda3\envs\eeg-notebooks_py38\lib\site-packages\pylsl\lib\liblsl64.dll' (or one of its dependencies). Try using the full path with constructor syntax.

danielemarinazzo, 2020-11-06:
sorry forgot 
OS:Windows 10 64 bit
device: Muse2, correctly streaming with bluemuse
I also performed the various additional suggested steps (ipykernel etc)

JohnGriffiths, 2020-11-06:
1. Start ipython in terminal

```bash
`ipython`
```

2.  Is `pylsl` installed?

```python
import pylsl
```

3. If so, is `muselsl` working?

```python
from muselsl import stream, list_muses, record
```

4. If so, is eeg-notebooks EEG class working?

```python
from eegnb.devices.eeg import EEG
```

JohnGriffiths, 2020-11-06:
Also, fyi: the  `run_notebooks.py` script will shortly be deprecated, as we have moved its functionality into a (much better) stand-alone command-line interface (CLI). 

To use that, run in anaconda terminal (not in python)

```bash
eegnb runexp -ip 
```

and follow the instructions

danielemarinazzo, 2020-11-09:
Hi,

thanks for your reply! 
Unfortunately it did not work, there is this issue when importing pylsl from python

```
(eeg-notebooks_py38) C:\Users\ricercasc\eeg-notebooks>ipython
Python 3.8.5 (default, Sep  3 2020, 21:29:08) [MSC v.1916 64 bit (AMD64)]
Type 'copyright', 'credits' or 'license' for more information
IPython 7.19.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import pylsl
---------------------------------------------------------------------------
FileNotFoundError                         Traceback (most recent call last)
<ipython-input-1-d9db44c5cf92> in <module>
----> 1 import pylsl

~\Miniconda3\envs\eeg-notebooks_py38\lib\site-packages\pylsl\__init__.py in <module>
      1 # (not using import * for Python 2.5 support)
----> 2 from .pylsl import IRREGULAR_RATE, DEDUCED_TIMESTAMP, FOREVER, cf_float32,\
      3     cf_double64, cf_string, cf_int32, cf_int16, cf_int8, cf_int64,\
      4     cf_undefined, protocol_version, library_version, library_info, local_clock,\
      5     proc_ALL, proc_none, proc_clocksync, proc_dejitter, proc_monotonize, proc_threadsafe,\

~\Miniconda3\envs\eeg-notebooks_py38\lib\site-packages\pylsl\pylsl.py in <module>
   1215                        "subfolder of the pylsl package or the system search "
   1216                        "path). Alternatively, specify the env var PYLSL_LIB")
-> 1217 lib = CDLL(libpath)
   1218
   1219 # set function return types where necessary

~\Miniconda3\envs\eeg-notebooks_py38\lib\ctypes\__init__.py in __init__(self, name, mode, handle, use_errno, use_last_error, winmode)
    379
    380         if handle is None:
--> 381             self._handle = _dlopen(self._name, mode)
    382         else:
    383             self._handle = handle

FileNotFoundError: Could not find module 'C:\Users\ricercasc\Miniconda3\envs\eeg-notebooks_py38\lib\site-packages\pylsl\lib\liblsl64.dll' (or one of its dependencies). Try using the full path with constructor syntax.
```

which could be related to this issue

https://stackoverflow.com/questions/59330863/cant-import-dll-module-in-python

I then tried with python 3.7, with a new environment, and now a slightly different issue popped up

(eeg-notebooks_py37) C:\Users\ricercasc\eeg-notebooks>ipython
Python 3.7.9 (default, Aug 31 2020, 17:10:11) [MSC v.1916 64 bit (AMD64)]
Type 'copyright', 'credits' or 'license' for more information
IPython 7.19.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import pylsl
---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
<ipython-input-1-d9db44c5cf92> in <module>
----> 1 import pylsl

c:\users\ricercasc\miniconda3\envs\eeg-notebooks_py37\lib\site-packages\pylsl\__init__.py in <module>
      1 # (not using import * for Python 2.5 support)
----> 2 from .pylsl import IRREGULAR_RATE, DEDUCED_TIMESTAMP, FOREVER, cf_float32,\
      3     cf_double64, cf_string, cf_int32, cf_int16, cf_int8, cf_int64,\
      4     cf_undefined, protocol_version, library_version, library_info, local_clock,\
      5     proc_ALL, proc_none, proc_clocksync, proc_dejitter, proc_monotonize, proc_threadsafe,\

c:\users\ricercasc\miniconda3\envs\eeg-notebooks_py37\lib\site-packages\pylsl\pylsl.py in <module>
   1215                        "subfolder of the pylsl package or the system search "
   1216                        "path). Alternatively, specify the env var PYLSL_LIB")
-> 1217 lib = CDLL(libpath)
   1218
   1219 # set function return types where necessary

c:\users\ricercasc\miniconda3\envs\eeg-notebooks_py37\lib\ctypes\__init__.py in __init__(self, name, mode, handle, use_errno, use_last_error)
    362
    363         if handle is None:
--> 364             self._handle = _dlopen(self._name, mode)
    365         else:
    366             self._handle = handle

OSError: [WinError 126] The specified module could not be found

JohnGriffiths, 2020-11-09:
Arg this is a challenging one. 

Ok, next:

Can you create fresh envs for python 3.7,3.8,3.9, and only install pylsl in those. Then activate and `import pylsl`. Does it ever work?

danielemarinazzo, 2020-11-09:
I did (including 3.6) and works in all of them :)
and now also the python run_notebooks.py works!

I had previously installed visual studio (following a suggestion on missing dll stuff), so I will never know if it was this step which solved the problem (probably yes), or having installed pylsl first, separately.

anyway, yay!

JohnGriffiths, 2020-11-09:
Ok , so you are working now then?

Good to hear
(although a bit confusing)

Next things for you then are 

a) view live data with 

`muselsl view -v 2`

And

b) try generating some erp data with n170 experiment

( see above comment - `run_notebooks.py` is now replaced by `eegnb runexp -ip` )

For that I recommend run multiple blocks (~15min total) of 2 min recordings. Then load in, following the `examples` code, and see if you get a notable difference at ~200ms

danielemarinazzo, 2020-11-09:
thanks a lot, I am doing this now (with the new command line interface).

on the other hand I have a kernel error when opening the viz/analysis notebook (gonna open a separate issue)

closed 2020-11-09 by danielemarinazzo