NeuroTechX/EEG-ExPy #23: kernel error - again dll

danielemarinazzo, 2020-11-09:
Hi

I am trying to run the SSVEP_load_and_visualize.ipynb notebook, and I get a kernel error (python 3.8, Windows 10), see below.

I tried the same on Ubuntu 18 and no error there

```
Traceback (most recent call last):
  File "c:\users\ricercasc\miniconda3\envs\eeg-notebooks_py38\lib\site-packages\tornado\web.py", line 1704, in _execute
    result = await result
  File "c:\users\ricercasc\miniconda3\envs\eeg-notebooks_py38\lib\site-packages\tornado\gen.py", line 769, in run
    yielded = self.gen.throw(*exc_info)  # type: ignore
  File "c:\users\ricercasc\miniconda3\envs\eeg-notebooks_py38\lib\site-packages\notebook\services\sessions\handlers.py", line 69, in post
    model = yield maybe_future(
  File "c:\users\ricercasc\miniconda3\envs\eeg-notebooks_py38\lib\site-packages\tornado\gen.py", line 762, in run
    value = future.result()
  File "c:\users\ricercasc\miniconda3\envs\eeg-notebooks_py38\lib\site-packages\tornado\gen.py", line 769, in run
    yielded = self.gen.throw(*exc_info)  # type: ignore
  File "c:\users\ricercasc\miniconda3\envs\eeg-notebooks_py38\lib\site-packages\notebook\services\sessions\sessionmanager.py", line 88, in create_session
    kernel_id = yield self.start_kernel_for_session(session_id, path, name, type, kernel_name)
  File "c:\users\ricercasc\miniconda3\envs\eeg-notebooks_py38\lib\site-packages\tornado\gen.py", line 762, in run
    value = future.result()
  File "c:\users\ricercasc\miniconda3\envs\eeg-notebooks_py38\lib\site-packages\tornado\gen.py", line 769, in run
    yielded = self.gen.throw(*exc_info)  # type: ignore
  File "c:\users\ricercasc\miniconda3\envs\eeg-notebooks_py38\lib\site-packages\notebook\services\sessions\sessionmanager.py", line 100, in start_kernel_for_session
    kernel_id = yield maybe_future(
  File "c:\users\ricercasc\miniconda3\envs\eeg-notebooks_py38\lib\site-packages\tornado\gen.py", line 762, in run
    value = future.result()
  File "c:\users\ricercasc\miniconda3\envs\eeg-notebooks_py38\lib\site-packages\notebook\services\kernels\kernelmanager.py", line 176, in start_kernel
    kernel_id = await maybe_future(self.pinned_superclass.start_kernel(self, **kwargs))
  File "c:\users\ricercasc\miniconda3\envs\eeg-notebooks_py38\lib\site-packages\jupyter_client\multikernelmanager.py", line 185, in start_kernel
    km.start_kernel(**kwargs)
  File "c:\users\ricercasc\miniconda3\envs\eeg-notebooks_py38\lib\site-packages\jupyter_client\manager.py", line 309, in start_kernel
    kernel_cmd, kw = self.pre_start_kernel(**kw)
  File "c:\users\ricercasc\miniconda3\envs\eeg-notebooks_py38\lib\site-packages\jupyter_client\manager.py", line 256, in pre_start_kernel
    self.write_connection_file()
  File "c:\users\ricercasc\miniconda3\envs\eeg-notebooks_py38\lib\site-packages\jupyter_client\connect.py", line 468, in write_connection_file
    self.connection_file, cfg = write_connection_file(self.connection_file,
  File "c:\users\ricercasc\miniconda3\envs\eeg-notebooks_py38\lib\site-packages\jupyter_client\connect.py", line 138, in write_connection_file
    with secure_write(fname) as f:
  File "c:\users\ricercasc\miniconda3\envs\eeg-notebooks_py38\lib\contextlib.py", line 113, in __enter__
    return next(self.gen)
  File "c:\users\ricercasc\miniconda3\envs\eeg-notebooks_py38\lib\site-packages\jupyter_core\paths.py", line 435, in secure_write
    win32_restrict_file_to_user(fname)
  File "c:\users\ricercasc\miniconda3\envs\eeg-notebooks_py38\lib\site-packages\jupyter_core\paths.py", line 361, in win32_restrict_file_to_user
    import win32api
ImportError: DLL load failed while importing win32api: The specified module could not be found.
```

JohnGriffiths, 2022-06-16:
Ciao @danielemarinazzo. Is this still an issue for you?

danielemarinazzo, 2022-06-16:
yes, I tried to install brainflow then uninstall it with pip, but eegnb still gives that error, whichever command I give (also not related to brainflow).

danielemarinazzo, 2022-06-16:
maybe, it was on another PC that I don't have at the moment. Gonna try in July but makes sense

tmorshed, 2024-04-04:
Is the issue still happening @danielemarinazzo?
If not, will close. Kindly let us know - thank you!

closed 2024-04-04 by tmorshed