NeuroTechX/EEG-ExPy #29: PyWinhook building wheel error

m9h, 2020-11-29:
On Win10 using Anaconda 4.8.4

conda create -n eeg-notebooks python=3.6 pip
conda activate eeg-notebooks
pip install -r requirements.txt

 Building wheel for pyWinhook (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: 'C:\Users\Morgan Hough\.conda\envs\eeg-notebooks\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\MORGAN~1\\AppData\\Local\\Temp\\pip-install-ntayfg3e\\pywinhook\\setup.py'"'"'; __file__='"'"'C:\\Users\\MORGAN~1\\AppData\\Local\\Temp\\pip-install-ntayfg3e\\pywinhook\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d 'C:\Users\MORGAN~1\AppData\Local\Temp\pip-wheel-yrqp9qeq'
       cwd: C:\Users\MORGAN~1\AppData\Local\Temp\pip-install-ntayfg3e\pywinhook\
  Complete output (16 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-3.6
  creating build\lib.win-amd64-3.6\pyWinhook
  copying pyWinhook\aa hook.py -> build\lib.win-amd64-3.6\pyWinhook
  copying pyWinhook\doc.py -> build\lib.win-amd64-3.6\pyWinhook
  copying pyWinhook\example.py -> build\lib.win-amd64-3.6\pyWinhook
  copying pyWinhook\HookManager.py -> build\lib.win-amd64-3.6\pyWinhook
  copying pyWinhook\__init__.py -> build\lib.win-amd64-3.6\pyWinhook
  running build_ext
  building 'pyWinhook._cpyHook' extension
  swigging pyWinhook/cpyHook.i to pyWinhook/cpyHook_wrap.c
  swig.exe -python -o pyWinhook/cpyHook_wrap.c pyWinhook/cpyHook.i
  error: command 'swig.exe' failed: No such file or directory
  ----------------------------------------
  ERROR: Failed building wheel for pyWinhook
  Running setup.py clean for pyWinhook

Need to use conda to install swig as there is no PyPI compatible package 

conda install swig

    running build_ext
    building 'pyWinhook._cpyHook' extension
    swigging pyWinhook/cpyHook.i to pyWinhook/cpyHook_wrap.c
    C:\Users\Morgan Hough\.conda\envs\eeg-notebooks\Library\bin\swig.exe -python -o pyWinhook/cpyHook_wrap.c pyWinhook/cpyHook.i
    error: Microsoft Visual C++ 14.0 is required. Get it with "Build Tools for Visual Studio": https://visualstudio.microsoft.com/downloads/
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'C:\Users\Morgan Hough\.conda\envs\eeg-notebooks\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\MORGAN~1\\AppData\\Local\\Temp\\pip-install-cb193sym\\pywinhook\\setup.py'"'"'; __file__='"'"'C:\\Users\\MORGAN~1\\AppData\\Local\\Temp\\pip-install-cb193sym\\pywinhook\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\MORGAN~1\AppData\Local\Temp\pip-record-hrg_h6jz\install-record.txt' --single-version-externally-managed --compile --install-headers 'C:\Users\Morgan Hough\.conda\envs\eeg-notebooks\Include\pyWinhook' Check the logs for full command output.

Now looking for the required Visual Studio build tools 

https://stackoverflow.com/questions/48541801/microsoft-visual-c-14-0-is-required-get-it-with-microsoft-visual-c-build-t

I assume this is because of the Python 3.6 requirement. Installing just the Desktop C++ build tools using Studio 17 solves the remaining issue building the wheel. 

Studio 19 is the current version and using a Python 3.7 requirement or forcing a PyWinhook==1.6.1 in the requirements.txt could be alternative workarounds.

JohnGriffiths, 2020-11-29:
Ok - can you re-try with pywinhook==1.6.1 ?

I found I had to add that last night after adding the docs dependencies into the requirements.txt. Maintaining separate docs and non docs requirements files seems to have been more work than just putting everything in one so far. 

I just ran with pywinhook==1.6.1 on windows and was ok.

JohnGriffiths, 2020-11-29:
maybe we should aim for python 3.7. Miniconda page does actually stay that's the latest stable version

https://docs.conda.io/projects/conda/en/latest/user-guide/install/download.html#anaconda-or-miniconda

JohnGriffiths, 2020-11-29:
Tried with a new miniconda install, python 3.7 env, and pywinhook 1.6.1.  Seems to be installing ok.

m9h, 2020-11-29:
Yep, I just confirmed that with one of the students that first reported the problem. python=3.7 env and it brings in pywinhook 1.6.0 when installing requirements.txt

JohnGriffiths, 2020-11-30:
Sorry not clear: are you saying all is good for python 3.7 at env creation & pywinhook 1.6.1 in requirements.txt

ErikBjare, 2020-12-02:
Got the same error when setting up the CI pipeline. You can install swig in a similar fashion to how we install it in CI:

https://github.com/NeuroTechX/eeg-notebooks/blob/d55110b763929f4f68885066ba345ed94f69aefc/.github/workflows/test.yml#L24-L29

You probably need to modify the `Out-File -FilePath $env:GITHUB_PATH -Encoding utf8 -Append` part, as that is specific to GitHub CI: https://stackoverflow.com/questions/60169752/how-to-update-the-path-in-a-github-action-workflow-file-for-a-windows-latest-hos/64874579#64874579

The ideal thing here would be to get rid of pywinhook altogether, but that might not be feasible. (I've had many problems with pyhook before in https://github.com/ActivityWatch/aw-watcher-afk)

ErikBjare, 2020-12-02:
I'm working on improving the situation for aw-watcher-afk in this PR: https://github.com/ActivityWatch/aw-watcher-afk/pull/44

If I get it working (which it seems like I will), then I will adapt the changes and make a PR.

JohnGriffiths, 2021-03-19:
Question for this: will all pywinhook issues be resolved by moving to newer python versions?

PR #32 (2021-03-26): fix: support Python 3.8 (and 3.9 on Linux)
Depends on #31

Fixes #29, Fixes #50 

Based on a similar solution in: https://github.com/ActivityWatch/aw-watcher-afk/pull/44
closed 2021-03-26 by ErikBjare