NeuroTechX/EEG-ExPy #34: Machine learning functionality - clustering

HussainAther, 2020-12-02:
Use signal processing metrics to create machine learning functionality for EEG-Notebooks and compare clustering algorithms. We can use scikit t-SNE optimization techniques (https://scikit-learn.org/stable/modules/manifold.html#optimizing-t-sne), Pyriemann Embedding (https://pyriemann.readthedocs.io/en/latest/auto_examples/ERP/plot_embedding_EEG.html#sphx-glr-auto-examples-erp-plot-embedding-eeg-py), diffusion_map (https://pydiffmap.readthedocs.io/en/master/reference/diffusion_map.html), and other ideas (anything else?)

Follow my work on Twitch: https://www.twitch.tv/hussainather

HussainAther, 2020-12-04:
- [ ] Show plots for different ML techniques (t-SNE, TangentSpace, Embedding). 
- [ ] Have presentation ready by 16:00 EST.

HussainAther, 2020-12-05:
Finishing up the `MLapproaches.ipynb` file to turn it into a page of the documentation for the NeurotechX repository.

- [ ] Add more background info on what this documentation is meant to be so the reader can understand.
- [ ] Get the content finalized and tidied. Rearrange the cells into an organized order and make sure each of them can run on both sets of data. 
- [ ] Export the file as a `.py` file for sphinx gallery. 
- [ ] Make a PR. Follow the style of other pull requests.

JohnGriffiths, 2020-12-05:
Awesome!

I suggest call the final file 

`examples/visual_n170/04r__n170_clustering.py`

HussainAther, 2020-12-05:
Thanks! Would you like me to add `pyDiffMap` to the `requirements.txt` file?

JohnGriffiths, 2021-03-19:
@HussainAther  I believe this is now summarized in the PR 

https://github.com/NeuroTechX/eeg-notebooks/pull/58

We'll review that and resume comments there :)