NeuroTechX/EEG-ExPy #37: Conda 1st time install conflicts with existing python install

retiutut, 2020-12-02:
I tried:

```
conda create -n "eeg-notebooks"

conda activate "eeg-notebooks"

conda install git

conda install pip

git clone https://github.com/NeuroTechX/eeg-notebooks

cd eeg-notebooks

pip install -r requirements.txt
```

but encountered errors on the last step, since it tried to use my existing python 2 install on my Mac.

@m9h helped me using a different set of instructions.

retiutut, 2020-12-03:
I found some instructions that seem similar to what helped me in a currently unused markdown file:

https://github.com/NeuroTechX/eeg-notebooks/blob/master/doc/getting_started/setup_overview.md#2-clone-eeg-notebooks-and-install-dependencies

ErikBjare, 2020-12-03:
This is very surprising to me. If you ran `conda activate "eeg-notebooks"` in the same shell as you later ran `pip` then shouldn't that override any system pip install by placing the other bin path earlier in the PATH? (that's at least how plain `venv` and similar behave)

Curious how that wasn't the case. What was the difference that fixed it for you?

retiutut, 2020-12-03:
@ErikBjare This only happened once, and I would like to try again with a fresh start on a different Mac that already has Python, but not Conda, installed.

I'll report back and close this issue if I am unable to replicate.

retiutut, 2020-12-03:
>Curious how that wasn't the case. What was the difference that fixed it for you?

```
conda create -n "eeg-notebooks" Python=3
```

This was the crucial step that made it work with no issues.

ErikBjare, 2020-12-03:
@retiutut Aaah, that makes a lot of sense. Surprised to see conda not pick Python 3 by default.

ErikBjare, 2020-12-04:
@retiutut I made a docs PR #51 that should help avoid similar issues in the future.

commit 985eb247 (2020-12-04): docs: improved installation instructions for conda (fixes #37)
commit c2035efa (2020-12-04): Merge pull request #51 from NeuroTechX/ErikBjare-patch-1

docs: improved installation instructions for conda (fixes #37)
closed 2020-12-04 by ErikBjare