NeuroTechX/EEG-ExPy #42: Docs Review - Audience

Learmstrong, 2020-12-03:
I am assuming the “audience” we are writing to includes educators at the High School or University level, in addition to students at these respective levels? How much background with these technologies do we expect our audience to have?

(Some of the pain points I am noting related to code or explication might be specific to my lack of experience, and unnecessary for our particular audience.)

JohnGriffiths, 2020-12-03:
It's a good question. 

Really to properly address multiple audiences we will need multiple documentation sets, pitched at different levels. 

It's probably a bit beyond the scope of the current docs to fully accommodate high school educators (we have actually done high school based projects with this tool; I know what's involved in that...)

But we certainly do motivated people with minimal computational background and minimal psychology+neuroscience knowledge to know what's going on. So there's lots of room for improvement with just that objective.

Learmstrong, 2020-12-04:
Hi John -

Thanks so much for the feedback. I did my best to provide comments that might strengthen this objective in the docs review - now located on [Issue 49](https://github.com/NeuroTechX/eeg-notebooks/issues/49). Appreciate you!

closed 2020-12-04 by Learmstrong