NeuroTechX/EEG-ExPy #43: Merge changes to the device abstraction ("EEG") class from my repo

ErikBjare, 2020-12-03:
My stuff is in: https://github.com/ErikBjare/thesis/blob/master/src/eegwatch/devices/eeg.py

### TODO

 - [ ] start/stop is different for Brainflow vs Muse devices, this should maybe be refactored, but at least be documented.
 - [ ] I've ran the black formatter on all my code, so merging would be easiest after similar formatting has been applied on the eeg-notebooks project (like in #30)

JadinTredup, 2020-12-15:
@ErikBjare it appears to me that you have completed this or is there still more work to be done?

ErikBjare, 2020-12-23:
@JadinTredup Nope, never got around to making a PR. 

I've made some pretty major changes, and working on a few more, almost to the point where I'm not sure if it makes sense to upstream it. But if it's of interest I'll consider it: https://github.com/ErikBjare/thesis/tree/master/src/eegwatch/devices

JohnGriffiths, 2021-03-19:
@ErikBjare  - what's your thoughts. Do you think these changes should be added or is divergence too great?

ErikBjare, 2021-04-11:
@JohnGriffiths I'm not sure, I'd love it if you took a look at it and told me your opinion on the matter.

Code is here: https://github.com/ErikBjare/thesis/tree/master/src/eegwatch/devices

Basically I created an abstract class `EEGDevice` that is implemented by the classes `MuseDevice` and `BrainflowDevice`. It simplifies a lot of the code imo, no more `if` statements to check if using one type of device or another, for example.

I also implemented a `check` method (only for MuseDevice so far) which does a very basic signal check (old method, not the newer one I linked you recently) and returns the list of bad channels, but that probably doesn't belong in the class (at least not in its current state).

JadinTredup, 2021-04-11:
@ErikBjare I am going to look this over today. Thank you so much for doing this. One of my biggest take aways from my internship development-wise has been implementing classes like this and so I was intending to do this at some point.

JadinTredup, 2021-04-11:
@JohnGriffiths I like this a lot and think it improves our flexibility/scalability in the future. If you're okay with it I say we set up another branch and work on merging this.

ErikBjare, 2021-04-13:
@JadinTredup Branch set up in #92, comments welcome!