NeuroTechX/EEG-ExPy #48: [Docs] Clarify Saved Data Location on Mac

retiutut, 2020-12-04:
Currently, this section of the docs exists but is blank. New users might not understand how to view hidden folders, but @JohnGriffiths explained that this is done as part of an existing convention.

Full Clarification for location on Mac:
```
You can find recording files for all experiments in a folder called .eegnb in the User's Home folder, as seen in the following screenshot.
```
[image]

retiutut, 2020-12-04:
I like the language already in place for the windows file location:

>The default directory is found at the location C:\Users\*USER_NAME*\.eegnb an example of which is pictured below.

retiutut, 2020-12-04:
On Mac, the default directory is found at the location `[HDD]/Users/Richard/.eegnb/data`, as seen in the picture below.
[Use Picture Attached to this Issue]

retiutut, 2021-03-05:
John suggests maybe just copying this data to the Documents folder.

It's still important for data to be stored in these hidden folders, as this is a standard used by applications.

JohnGriffiths, 2021-03-19:
@retiutut - do you need any help / pointers / discussion in implementing this?

retiutut, 2021-04-12:
I think there is enough info here to implement a solution.

closed 2026-08-30 by pellet