NeuroTechX/EEG-ExPy #50: Support newer Python versions

ErikBjare, 2020-12-04:
There seems to be a lot of issues when installing on other Python versions than 3.7, since I have a pretty extensive history of battling issues like these I thought I might give it a shot (especially now that we have a CI pipeline).

Documenting known issues here:

 - [ ] Installing pywinhook on Windows on Python 3.9 requires extra build deps (swig, VS C++ tools)
 - [ ] Build errors for numpy/pandas

JohnGriffiths, 2021-03-19:
We'll aim to address newer Python version support with high priority. 

However, it is quite tricky to make quick progress on as we need to co-ordinate multiple people with access to different OSs etc. 
(as we observed at the Dec brainhack)

JohnGriffiths, 2021-03-19:
Related: https://github.com/NeuroTechX/eeg-notebooks/issues/29

PR #32 (2021-03-26): fix: support Python 3.8 (and 3.9 on Linux)
Depends on #31

Fixes #29, Fixes #50 

Based on a similar solution in: https://github.com/ActivityWatch/aw-watcher-afk/pull/44
closed 2021-03-26 by ErikBjare