NeuroTechX/EEG-ExPy #66: sounds fades away in ssaep experiments

danielemarinazzo, 2020-12-14:
Hi

in both ssaep experiments the sounds fades away after a few instances, becoming barely audible.
maybe it's my perception but it's somehow less evident in the two frequencies one (4a), but still very real.

danielemarinazzo, 2021-01-13:
update, it works now, I installed manually pyo and pygame, and put one of these two as default

PTB still gave an error

`self.stopTime = self._nSamples / float(self.sampleRate) TypeError: float() argument must be a string or a number, not 'NoneType'`

I also changed the output sound from "native sound" or something like that, to another option, with four capital letters.

Why am I not being more specific? Because now psychopy gives an error when I try to access the preferences :) 

```
Traceback (most recent call last):
  File "c:\users\daniele\miniconda3\envs\eegnb_py37\lib\site-packages\psychopy\app\_psychopyApp.py", line 767, in showPrefs
    prefsDlg = PreferencesDlg(app=self)
  File "c:\users\daniele\miniconda3\envs\eegnb_py37\lib\site-packages\psychopy\app\preferencesDlg.py", line 543, in __init__
    devnames = sorted(sound.getDevices('output'))
AttributeError: module 'psychopy.sound' has no attribute 'getDevices'
```

But as long as the sound works, I won't touch it :)

JohnGriffiths, 2021-02-20:
@danielemarinazzo  - try changing the SOA ( = the trial duration ). 

Tried this just now.

With SOA <= 1 I get the volume dropoff, and it's worse for e.g. SOA=0.5

At SOA = 2.0 I didn't hear any volume dropoff over 1 min. 

~~~

Also note here: you could set longer SOAs and have fewer, longer trials. That would remove the unfortunate clicks that accompany the stim onset offset (which also needs fixing). 

Since you set ITI to zero for your `ssaep_onefreq` experiment, I guess what you are after is a single continuous sound stream. 

Note that there might be advantages to having delineated trials at later points for averaging etc; my first thought would be maybe 30s trials, for a 5-10 min expt.

JohnGriffiths, 2021-02-20:
Ok sorry it still fades out for longer runs. So must be to do with the number of repeats.

JohnGriffiths, 2021-02-20:
Still, with 30s trials you should have no issues...

JohnGriffiths, 2021-02-20:
Ok, have a better fix: move the sound object creation inside the trials loop. No fading with this. Doing PR now.

JohnGriffiths, 2021-02-20:
Here https://github.com/NeuroTechX/eeg-notebooks/pull/81

danielemarinazzo, 2021-02-21:
thanks a lot! what do you have as psychopy sound preferences? Because for me what resolved the issue was to install pyo and pygame nad change the audio type in psychopy.

btw do you manage to use PTB as primary sound preference?

danielemarinazzo, 2022-06-16:
Last time I checked was in April, with a fresh installation of the toolbox at that time, on a different PC (still windows, still python 3.7). Now I cannot check since my new installation does not work

danielemarinazzo, 2022-06-16:
So the installation works now, and the sound does not fade. I cannot check if it's PTB or else since when I run the `psychopy` command it does not start. Any ideas? In any case it works and psychopy does not complain about using the wrong sound options.