NeuroTechX/EEG-ExPy #68: error after changing psychopy PTB sound preferences

danielemarinazzo, 2020-12-14:
So there's this warning

`3.6146  WARNING         We strongly recommend you activate the PTB sound engine in PsychoPy prefs as the preferred audio engine. Its timing is vastly superior. Your prefs are currently set to use ['sounddevice', 'PTB', 'pyo', 'pygame'] (in that order).`

If I do as requested, I then get an error

```
Traceback (most recent call last):
  File "C:\Users\daniele\miniconda3\envs\eegnb_py37\Scripts\eegnb-script.py", line 33, in <module>
    sys.exit(load_entry_point('eeg-notebooks', 'console_scripts', 'eegnb')())
  File "c:\users\daniele\dropbox\code\eeg-notebooks\eegnb\cli\__main__.py", line 74, in main
    cli = CLI(args.command)
  File "c:\users\daniele\dropbox\code\eeg-notebooks\eegnb\cli\cli.py", line 9, in __init__
    getattr(self, command)()
  File "c:\users\daniele\dropbox\code\eeg-notebooks\eegnb\cli\cli.py", line 70, in runexp
    run_introprompt()
  File "c:\users\daniele\dropbox\code\eeg-notebooks\eegnb\cli\introprompt.py", line 155, in main
    run_experiment(experiment, record_duration, eeg_device, save_fn)
  File "c:\users\daniele\dropbox\code\eeg-notebooks\eegnb\cli\utils.py", line 62, in run_experiment
    ssaep_onefreq.present(duration=record_duration, eeg=eeg_device, save_fn=save_fn)
  File "c:\users\daniele\dropbox\code\eeg-notebooks\eegnb\experiments\auditory_ssaep\ssaep_onefreq.py", line 104, in present
    aud1 = sound.Sound(am1)
  File "C:\Users\daniele\miniconda3\envs\eegnb_py37\lib\site-packages\psychopy\sound\backend_ptb.py", line 334, in __init__
    hamming=self.hamming)
  File "C:\Users\daniele\miniconda3\envs\eegnb_py37\lib\site-packages\psychopy\sound\backend_ptb.py", line 413, in setSound
    _SoundBase.setSound(self, value, secs, octave, hamming, log)
  File "C:\Users\daniele\miniconda3\envs\eegnb_py37\lib\site-packages\psychopy\sound\_base.py", line 198, in setSound
    self._setSndFromArray(numpy.array(value))
  File "C:\Users\daniele\miniconda3\envs\eegnb_py37\lib\site-packages\psychopy\sound\backend_ptb.py", line 479, in _setSndFromArray
    self.stopTime = self._nSamples / float(self.sampleRate)
TypeError: float() argument must be a string or a number, not 'NoneType'
```

JohnGriffiths, 2020-12-25:
Thanks. Will look into this. I also looked into this unsystematically when tidying up the auditory experiments, and found different issues with PTB (additional installation requirements, and apparently worse sound quality). This is a different problem, but in any case we should review the PTB option properly, as psychopy does seem to have a major preference for it.

andraderenew, 2021-05-20:
I have tried to run like in this code I add here. But it seems that this issue has been presented before. 

(eeg-notebooks) andraderenew@Renes-MacBook-Air eeg-notebooks-master % eegnb runexp -ip
Experiment to run: 1
run command line prompt script
Welcome to NeurotechX EEG Notebooks

Please enter the integer value corresponding to your EEG device: 

[ 0] None
[ 1] Muse (2016)
[ 2] Muse 2
[ 3] Muse S
[ 4] OpenBCI Ganglion
[ 5] OpenBCI Cyton
[ 6] OpenBCI Cyton + Daisy
[ 7] G.Tec Unicorn
[ 8] BrainBit
[ 9] Notion 1
[10] Notion 2
[11] Synthetic
[12] FreeEEG32

Enter Board Selection: 2
Selected board: Muse 2 

Please select which experiment you would like to run: 

[0] Visual N170
[1] Visual P300
[2] Visual SSVEP
[3] visual-cue (no description)
[4] visual-codeprose (no description)
[5] Auditory SSAEP (orig)
[6] Auditory SSAEP (single freq)
[7] Auditory oddball (orig)
[8] Auditory oddball (diaconescu)

Enter Experiment Selection: 0
Selected experiment: visual-N170 

Now, enter the duration of the recording (in seconds). 

Enter duration: 20

Next, enter the ID# of the subject you are recording data from. 

Enter subject ID#: 1

Next, enter the session number you are recording for. 

Enter session #: 1

EEG device successfully connected!
Press [ENTER] when ready to begin...
Traceback (most recent call last):
  File "/Users/andraderenew/opt/anaconda3/bin/eegnb", line 33, in <module>
    sys.exit(load_entry_point('eeg-notebooks', 'console_scripts', 'eegnb')())
  File "/Users/andraderenew/opt/anaconda3/lib/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/Users/andraderenew/opt/anaconda3/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/Users/andraderenew/opt/anaconda3/lib/python3.8/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/andraderenew/opt/anaconda3/lib/python3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/andraderenew/opt/anaconda3/lib/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/Users/andraderenew/eeg-notebooks/eegnb/cli/__main__.py", line 55, in runexp
    run_introprompt()
  File "/Users/andraderenew/eeg-notebooks/eegnb/cli/introprompt.py", line 132, in main
    run_experiment(experiment, record_duration, eeg_device, save_fn)
  File "/Users/andraderenew/eeg-notebooks/eegnb/cli/utils.py", line 38, in run_experiment
    module.present(duration=record_duration, eeg=eeg_device, save_fn=save_fn)  # type: ignore
  File "/Users/andraderenew/eeg-notebooks/eegnb/experiments/visual_n170/n170.py", line 22, in present
    record_duration = np.float32(duration)
TypeError: float() argument must be a string or a number, not 'EEG'
1.3636 	WARNING 	PTB audio lib was requested but not loaded: psychtoolbox audio failed to import
1.4181 	WARNING 	For experiments that use audio stimuli, timing will be much better if you upgrade your PsychoPy installation to a 64bit Python3 installation and use the PTB backend.

ErikBjare, 2021-05-21:
@andraderenew As said before in https://github.com/NeuroTechX/eeg-notebooks/issues/16#issuecomment-843180077, this issue was fixed in #101. Make sure you're using the latest commit.

andraderenew, 2021-05-21:
I am using the latest commit. I deleted all I had previously and reinstalled from source.

danielemarinazzo, 2021-05-21:
If the problem still persists, try to run psychopy before eegnb. Once you are in psychopy go to preferences->audio->audio library and puy pygame or pyo on top (you should have these installed), and leave ptb and sounddevice somewhere lower in the queue.

danielemarinazzo, 2021-05-21:
see also https://github.com/NeuroTechX/eeg-notebooks/issues/66#issuecomment-759330579 and replies

ErikBjare, 2021-05-21:
@danielemarinazzo I can see from the stack trace that you're not using the latest commit. You may have files from the old commit still around.

danielemarinazzo, 2021-05-21:
thanks for checking, I'll use the last commit!
I was pointing out a solution that worked for me when everything else did
not, in case @andraderenew is actually using the last commit.

ErikBjare, 2021-05-21:
@danielemarinazzo Sorry I tagged the wrong person, I meant to tag @andraderenew.

andraderenew, 2021-05-21:
all solved!!! thanks for the comment on PTB and psychopy. I had to set preferences with python console because with the GUI it crashed.

closed 2022-06-16 by JohnGriffiths