NeuroTechX/EEG-ExPy #70: Implement code vs prose task

ErikBjare, 2020-12-16:
As part of my [thesis](https://github.com/ErikBjare/thesis) work, I want to replicate (and outperform) a previous study (https://ieeexplore.ieee.org/abstract/document/8813259) by @dfucci et al that attempted to classify whether the subject is reading code or prose. Their classifier exhibited very modest performance, no doubt in part due to the limited 1-channel EEG equipment they were using, which I'm confident I can improve upon for my thesis.

The study by Fucci et al is in turn a replication study of an fMRI study (https://ieeexplore.ieee.org/abstract/document/7985660), which clearly showed there were differences in reading code and prose.

More recently (yesterday), MIT researchers published an fMRI study on the differences in brain activity between reading code and prose: https://news.mit.edu/2020/brain-reading-computer-code-1215 (the actual paper: https://elifesciences.org/articles/58906)

I think this task would be a nice addition to the notebooks, and I'm creating this issue to signal my intent to implement it :slightly_smiling_face:

JohnGriffiths, 2020-12-16:
Excellent!

Currently the Cueing experiment from Kyle is the most psychopy-sophisticated one in the repo, so I would suggest taking a look at that if you haven't already for more complex designs.

ErikBjare, 2020-12-16:
@JohnGriffiths Good to know, thanks! Been digging around and familiarizing myself with the experiment code, but functions that are hundreds of lines are a bit difficult to digest... :slightly_smiling_face: 

It led me to create #71 though, which should make it easier to add new experiments.

closed 2021-04-18 by ErikBjare