NeuroTechX/EEG-ExPy #77: Add code coverage analysis

ErikBjare, 2020-12-24:
Another improvement to the CI pipeline.

We don't have much in the way of tests, but we do have the auto examples which run a fair bit of the code. It'd be nice to know exactly which code is somewhat tested in CI, and which isn't (and when new code is added, how well covered it is by tests/examples). 

This should be easy enough to do with `coverage.py`.

JohnGriffiths, 2020-12-25:
Agree- the examples have a very test-like function but that isn't explicitly documented. Coverage tracking would be great to include

ErikBjare, 2021-03-26:
I added coverage for `*.ipynb` notebooks in my thesis repo the other day, but I'm not sure how to do it for the auto_examples... 

I gave it a shot in https://github.com/NeuroTechX/eeg-notebooks/pull/88, but it doesn't really work yet.

PR #88 (2021-05-20): ci: added minimal testing setup with coverage
fixes #77 

This simply runs all the examples with pytest + pytest-cov (for coverage) + nbval (for `.ipynb` files).

All the `*.py` files in the example folder are ignored since I'm not sure how to deal with the test collection when the entire file is the test (and may do things that require interaction). 

Since most of the notebooks seem to not run correctly I think it might make sense to try and figure out how to run the auto_examples as tests, but I can't figure out a way for pytest to treat each file like a test without running it at collectio [... cut]
closed 2021-05-20 by ErikBjare