NeuroTechX/EEG-ExPy #78: startup error

danielemarinazzo, 2021-01-06:
Hi

after updating my local repository, I am now no longer able to start the CLI. I have a python 3.7 virtual environment on Windows 10.

```
(eegnb_py37) C:\Users\daniele\Dropbox\code\eeg-notebooks>eegnb runexp -ip
run command line prompt script
Traceback (most recent call last):
  File "C:\Users\daniele\miniconda3\envs\eegnb_py37\Scripts\eegnb-script.py", line 33, in <module>
    sys.exit(load_entry_point('eeg-notebooks', 'console_scripts', 'eegnb')())
  File "c:\users\daniele\dropbox\code\eeg-notebooks\eegnb\cli\__main__.py", line 74, in main
    cli = CLI(args.command)
  File "c:\users\daniele\dropbox\code\eeg-notebooks\eegnb\cli\cli.py", line 9, in __init__
    getattr(self, command)()
  File "c:\users\daniele\dropbox\code\eeg-notebooks\eegnb\cli\cli.py", line 68, in runexp
    from .introprompt import main as run_introprompt
  File "c:\users\daniele\dropbox\code\eeg-notebooks\eegnb\cli\introprompt.py", line 4, in <module>
    from eegnb.devices.eeg import EEG
  File "c:\users\daniele\dropbox\code\eeg-notebooks\eegnb\devices\eeg.py", line 21, in <module>
    from eegnb.devices.utils import get_openbci_usb, create_stim_array
  File "c:\users\daniele\dropbox\code\eeg-notebooks\eegnb\devices\utils.py", line 41, in <module>
    "freeeeg32": BoardShim.get_eeg_channels(BoardIds.FREEEEG32_BOARD.value),
  File "C:\Users\daniele\miniconda3\envs\eegnb_py37\lib\enum.py", line 354, in __getattr__
    raise AttributeError(name) from None
AttributeError: FREEEEG32_BOARD
```

JohnGriffiths, 2021-01-06:
Hi Daniele. 

Sorry about this. 

This relates to @JadinTredup's [updating of FreeEEG board support](https://github.com/NeuroTechX/eeg-notebooks/blob/master/eegnb/devices/utils.py#L41)

First: could you check your brainflow version matches that in the requirements?

Second: check whether that is the latest brainflow version?

danielemarinazzo, 2021-01-06:
Thanks!

I tried to install the required version (which is not the latest one), but got a message of conflict with eeg-notebooks (which if I understood correctly should update automatically, if I installed it with -e.

```
(eegnb_py37) C:\Users\daniele>pip install brainflow==3.7.2
Collecting brainflow==3.7.2
  Downloading brainflow-3.7.2-py3-none-any.whl (17.6 MB)
     |████████████████████████████████| 17.6 MB 6.4 MB/s
Requirement already satisfied: nptyping in c:\users\daniele\miniconda3\envs\eegnb_py37\lib\site-packages (from brainflow==3.7.2) (1.3.0)
Requirement already satisfied: numpy in c:\users\daniele\miniconda3\envs\eegnb_py37\lib\site-packages (from brainflow==3.7.2) (1.19.1)
Requirement already satisfied: numpy in c:\users\daniele\miniconda3\envs\eegnb_py37\lib\site-packages (from brainflow==3.7.2) (1.19.1)
Requirement already satisfied: typish>=1.5.2 in c:\users\daniele\miniconda3\envs\eegnb_py37\lib\site-packages (from nptyping->brainflow==3.7.2) (1.9.1)
Installing collected packages: brainflow
  Attempting uninstall: brainflow
    Found existing installation: brainflow 3.4.3
    Uninstalling brainflow-3.4.3:
      Successfully uninstalled brainflow-3.4.3
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
eeg-notebooks 0.2 requires brainflow==3.4.3, but you have brainflow 3.7.2 which is incompatible.
Successfully installed brainflow-3.7.2
```

If I try to run eegnb runexp anyway I got an error message, requesting the 3.4.3 version

```
(eegnb_py37) C:\Users\daniele\Dropbox\code\eeg-notebooks>eegnb runexp -ip
run command line prompt script
Traceback (most recent call last):
  File "C:\Users\daniele\miniconda3\envs\eegnb_py37\lib\site-packages\pkg_resources\__init__.py", line 567, in _build_master
    ws.require(__requires__)
  File "C:\Users\daniele\miniconda3\envs\eegnb_py37\lib\site-packages\pkg_resources\__init__.py", line 884, in require
    needed = self.resolve(parse_requirements(requirements))
  File "C:\Users\daniele\miniconda3\envs\eegnb_py37\lib\site-packages\pkg_resources\__init__.py", line 775, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (brainflow 3.7.2 (c:\users\daniele\miniconda3\envs\eegnb_py37\lib\site-packages), Requirement.parse('brainflow==3.4.3'), {'eeg-notebooks'})

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\daniele\miniconda3\envs\eegnb_py37\Scripts\eegnb-script.py", line 33, in <module>
    sys.exit(load_entry_point('eeg-notebooks', 'console_scripts', 'eegnb')())
  File "c:\users\daniele\dropbox\code\eeg-notebooks\eegnb\cli\__main__.py", line 74, in main
    cli = CLI(args.command)
  File "c:\users\daniele\dropbox\code\eeg-notebooks\eegnb\cli\cli.py", line 9, in __init__
    getattr(self, command)()
  File "c:\users\daniele\dropbox\code\eeg-notebooks\eegnb\cli\cli.py", line 68, in runexp
    from .introprompt import main as run_introprompt
  File "c:\users\daniele\dropbox\code\eeg-notebooks\eegnb\cli\introprompt.py", line 4, in <module>
    from eegnb.devices.eeg import EEG
  File "c:\users\daniele\dropbox\code\eeg-notebooks\eegnb\devices\eeg.py", line 17, in <module>
    from brainflow import BoardShim, BoardIds, BrainFlowInputParams
  File "C:\Users\daniele\miniconda3\envs\eegnb_py37\lib\site-packages\brainflow\__init__.py", line 1, in <module>
    from brainflow.board_shim import *
  File "C:\Users\daniele\miniconda3\envs\eegnb_py37\lib\site-packages\brainflow\board_shim.py", line 4, in <module>
    import pkg_resources
  File "C:\Users\daniele\miniconda3\envs\eegnb_py37\lib\site-packages\pkg_resources\__init__.py", line 3238, in <module>
    @_call_aside
  File "C:\Users\daniele\miniconda3\envs\eegnb_py37\lib\site-packages\pkg_resources\__init__.py", line 3222, in _call_aside
    f(*args, **kwargs)
  File "C:\Users\daniele\miniconda3\envs\eegnb_py37\lib\site-packages\pkg_resources\__init__.py", line 3251, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "C:\Users\daniele\miniconda3\envs\eegnb_py37\lib\site-packages\pkg_resources\__init__.py", line 569, in _build_master
    return cls._build_from_requirements(__requires__)
  File "C:\Users\daniele\miniconda3\envs\eegnb_py37\lib\site-packages\pkg_resources\__init__.py", line 582, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "C:\Users\daniele\miniconda3\envs\eegnb_py37\lib\site-packages\pkg_resources\__init__.py", line 770, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'brainflow==3.4.3' distribution was not found and is required by eeg-notebooks

```

danielemarinazzo, 2021-01-06:
Update, I reinstalled eegnb manually after installing brainflow 3.7.2 and it works

closed 2021-01-06 by danielemarinazzo