NeuroTechX/EEG-ExPy #79: Pre-processing pipeline for low density EEG system

akshaysravindran, 2021-01-25:
The attached code proposes one prospective preprocessing pipeline that helps handle artifacts in EEG data collected using the Muse system. The text contains some additional preprocessing steps which  I typically use when working with denser EEG systems but some of these steps are not applicable to Muse. The code will focus on the sections without an asterisk.

 0)  Load the required libraries
 1)  Load the EEG, chanlocs into EEGlab format
 ii) Downsample if needed
 iii) High Pass filter (0.1-2 Hz: Depending on noise level & study)

 iv) *** H-infinity adaptive ocular artifact removal ***  (Note: not applicable for Muse and not included)

 v)  Low pass filter (depending on the study)
 vi) Remove Bad segments (Visually inspecting)
 vii) Remove Bad channels (Visual Inspection with Muse)
 Viii) Remove burst artifacts using ASR 

 ix) *** ICA cleaning *** (Note: not applicable for Muse and not included)
 x) *** Interpolating removed channels *** (Note: not applicable for Muse and not included)

 Additional Commands
 xi) Notch filter  (depending on the study)
 xii) Common Average Reference (Note: After ICA or add averaged channel if doing before ICA to account for rank deficiency)
 xiii) Multiple plots for sanity checks

[Pre_processing_tutorial.zip](https://github.com/NeuroTechX/eeg-notebooks/files/5868327/Pre_processing_tutorial.zip)

JadinTredup, 2021-02-13:
Is there equivalent Python code, or only Matlab?

JohnGriffiths, 2021-03-19:
This is really great and very important going forward for us to establish a more developed preprocessing pipeline. 

(Which unfortunately will often need to be somewhat device-specific)

Let's evaluate these new code, and consider alongside other preprocessing options, e.g.

https://github.com/mne-tools/mne-study-template/tree/master/scripts/preprocessing

Thanks!