NeuroTechX/EEG-ExPy #84: Loading FreeEEG32 recorded data

Div12345, 2021-03-08:
Hi, 
I ran an ssvep experiment using eeg-notebooks for ssvep and then I loaded it by modifying the ssvep visualisation example slightly. But while loading I seem to be having a problem.
The recording was from a 2 channel (FP1 and FP2) headset with the FreeEEG32 board doing the acquisition.
This was the code - 
```
eegnb_data_path = os.path.join(os.path.expanduser('~/'),'.eegnb', 'data')    
ssvep_data_path = os.path.join(eegnb_data_path, 'visual-SSVEP', 'local','freeeeg32')

subject = 1
session = 1
raw = load_data(subject, session, 
                experiment='visual-SSVEP', site='local', device_name='freeeeg32',
                data_dir = eegnb_data_path)
```

But this is the error I get - 
```
['eeg_0', 'eeg_1', 'eeg_2', 'eeg_3', 'eeg_4', 'eeg_5', 'eeg_6', 'eeg_7', 'eeg_8', 'eeg_9', 'eeg_10', 'eeg_11', 'eeg_12', 'eeg_13', 'eeg_14', 'eeg_15', 'eeg_16', 'eeg_17', 'eeg_18', 'eeg_19', 'eeg_20', 'eeg_21', 'eeg_22', 'eeg_23', 'eeg_24', 'eeg_25', 'eeg_26', 'eeg_27', 'eeg_28', 'eeg_29', 'eeg_30', 'eeg_31', 'stim']
Creating RawArray with float64 data, n_channels=33, n_times=31428
    Range : 0 ... 31427 =      0.000 ...    61.381 secs
Ready.

---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-4-f9984a7bc1ef> in <module>
     11 raw = load_data(subject, session, 
     12                 experiment='visual-SSVEP', site='local', device_name='freeeeg32',
---> 13                 data_dir = eegnb_data_path)

d:\softwares\github\eeg-notebooks\eegnb\analysis\utils.py in load_data(subject_id, session_nb, device_name, experiment, replace_ch_names, verbose, site, data_dir)
    164             ch_ind=ch_ind,
    165             replace_ch_names=replace_ch_names,
--> 166             verbose=verbose,
    167         )
    168 

d:\softwares\github\eeg-notebooks\eegnb\analysis\utils.py in load_csv_as_raw(fnames, sfreq, ch_ind, aux_ind, replace_ch_names, verbose)
     77     raws = concatenate_raws(raw, verbose=verbose)
     78     montage = make_standard_montage("standard_1005")
---> 79     raws.set_montage(montage)
     80 
     81     return raws

<decorator-gen-22> in set_montage(self, montage, match_case, on_missing, verbose)

d:\softwares\anaconda3\envs\eeg-notebooks\lib\site-packages\mne\io\meas_info.py in set_montage(self, montage, match_case, on_missing, verbose)
    160         from ..channels.montage import _set_montage
    161         info = self if isinstance(self, Info) else self.info
--> 162         _set_montage(info, montage, match_case, on_missing)
    163         return self
    164 

d:\softwares\anaconda3\envs\eeg-notebooks\lib\site-packages\mne\channels\montage.py in _set_montage(info, montage, match_case, on_missing)
    772                 'in your analyses.'
    773             )
--> 774             _on_missing(on_missing, missing_coord_msg)
    775 
    776             # set ch coordinates and names from digmontage or nan coords

d:\softwares\anaconda3\envs\eeg-notebooks\lib\site-packages\mne\utils\check.py in _on_missing(on_missing, msg, name)
    730     _check_option(name, on_missing, ['raise', 'warn', 'ignore'])
    731     if on_missing == 'raise':
--> 732         raise ValueError(msg)
    733     elif on_missing == 'warn':
    734         warn(msg)

ValueError: DigMontage is only a subset of info. There are 32 channel positions not present in the DigMontage. The required channels are:

['eeg_0', 'eeg_1', 'eeg_2', 'eeg_3', 'eeg_4', 'eeg_5', 'eeg_6', 'eeg_7', 'eeg_8', 'eeg_9', 'eeg_10', 'eeg_11', 'eeg_12', 'eeg_13', 'eeg_14', 'eeg_15', 'eeg_16', 'eeg_17', 'eeg_18', 'eeg_19', 'eeg_20', 'eeg_21', 'eeg_22', 'eeg_23', 'eeg_24', 'eeg_25', 'eeg_26', 'eeg_27', 'eeg_28', 'eeg_29', 'eeg_30', 'eeg_31'].

Consider using inst.set_channel_types if these are not EEG channels, or use the on_missing parameter if the channel positions are allowed to be unknown in your analyses.

```

What is the change I must do to get over this problem?
Thanks in Advance!

Div12345, 2021-03-08:
Also, is it possible to choose what channels to record/save/load? Or should I raise this as a separate issue?

JohnGriffiths, 2021-03-08:
Hi Divyesh. 

This looks solvable. 

Could you send me the eeg file you're trying to load and I'll take a look. 

@m9h - have you made any recordings with this board yet?

JohnGriffiths, 2021-03-19:
@Div12345 - thanks for the PR. We will review shortly and leave feedback there!

closed 2021-04-11 by JadinTredup