NeuroTechX/EEG-ExPy #96: Error with Setting Anaconda Environment

mikaelhaji, 2021-04-24:
- MacBook Pro 1 inch 2019 - Big Sur
- OpenBCI Cyton and Ultra Cortex 4

I was planning on testing out the P300 and SSVEP Stimuli. However, when setting up by Anaconda Environment, I kept on getting an error on "dukpy lib." I tried to pip install dukpy but it still doesn't work. I also tried binder but got another error -- from my understanding it hasn't been updated in a while anyway.

Any Help Would Be Greatly Appreciated!

## 📷 Screenshot
[image]

JohnGriffiths, 2021-04-24:
Hi Mikael. 

Let's try to get to the bottom of this. 

What do you get when you do `pip install dukpy` ?

mikaelhaji, 2021-04-24:
[image: Screen Shot 2021-04-24 at 6.55.17 PM.png]

Hey John,

Above is the error I am getting.

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
04/24/21,
06:55:22 PM

mikaelhaji, 2021-04-24:
^When I pip install dukpy

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
04/24/21,
06:57:41 PM

JohnGriffiths, 2021-04-25:
sorry - were you intending to attach a second screenshot there? 

or are you saying you get the same error when you try to pip install `dukpy` ?

mikaelhaji, 2021-04-25:
I get the exact same error. The second image was basically the same as the
first one.

mikaelhaji, 2021-04-26:
Just want to follow up on this error.

ErikBjare, 2021-04-26:
You need to provide full log output for us to be able to help you. 

Preferably in text form and not as a screenshot (at least not a screenshot sent through your mail client, since that doesn't seem to be accepted by GitHub).

mikaelhaji, 2021-04-26:
Sure, here it is.

WARNING: Value for scheme.headers does not match. Please report this to <
https://github.com/pypa/pip/issues/9617>

distutils:
/Library/Frameworks/Python.framework/Versions/3.8/include/python3.8/UNKNOWN

sysconfig:
/Library/Frameworks/Python.framework/Versions/3.8/include/python3.8

WARNING: Additional context:

user = False

home = None

root = None

prefix = None

Collecting dukpy

  Using cached dukpy-0.2.3.tar.gz (1.9 MB)

Using legacy 'setup.py install' for dukpy, since package 'wheel' is not
installed.

Installing collected packages: dukpy

    Running setup.py install for dukpy ... error

    ERROR: Command errored out with exit status 1:

     command:
/Library/Frameworks/Python.framework/Versions/3.8/bin/python3.8 -u -c
'import io, os, sys, setuptools, tokenize; sys.argv[0] =
'"'"'/private/var/folders/9s/3hlz7cm91n7975n552pg6fh40000gn/T/pip-install-ulu9kzcd/dukpy_a71c397eaff2464696eb4d0fdd56caab/setup.py'"'"';
__file__='"'"'/private/var/folders/9s/3hlz7cm91n7975n552pg6fh40000gn/T/pip-install-ulu9kzcd/dukpy_a71c397eaff2464696eb4d0fdd56caab/setup.py'"'"';f
= getattr(tokenize, '"'"'open'"'"', open)(__file__) if
os.path.exists(__file__) else io.StringIO('"'"'from setuptools import
setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"',
'"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))'
install --record
/private/var/folders/9s/3hlz7cm91n7975n552pg6fh40000gn/T/pip-record-g3mqsqim/install-record.txt
--single-version-externally-managed --compile --install-headers
/Library/Frameworks/Python.framework/Versions/3.8/include/python3.8/dukpy

         cwd:
/private/var/folders/9s/3hlz7cm91n7975n552pg6fh40000gn/T/pip-install-ulu9kzcd/dukpy_a71c397eaff2464696eb4d0fdd56caab/

    Complete output (145 lines):

    running install

    running build

    running build_py

    creating build

    creating build/lib.macosx-10.9-x86_64-3.8

    creating build/lib.macosx-10.9-x86_64-3.8/dukpy

    copying dukpy/run.py -> build/lib.macosx-10.9-x86_64-3.8/dukpy

    copying dukpy/__init__.py -> build/lib.macosx-10.9-x86_64-3.8/dukpy

    copying dukpy/lessc.py -> build/lib.macosx-10.9-x86_64-3.8/dukpy
[... 394 lines cut ...]
    ----------------------------------------

ERROR: Command errored out with exit status 1:
/Library/Frameworks/Python.framework/Versions/3.8/bin/python3.8 -u -c
'import io, os, sys, setuptools, tokenize; sys.argv[0] =
'"'"'/private/var/folders/9s/3hlz7cm91n7975n552pg6fh40000gn/T/pip-install-ulu9kzcd/dukpy_a71c397eaff2464696eb4d0fdd56caab/setup.py'"'"';
__file__='"'"'/private/var/folders/9s/3hlz7cm91n7975n552pg6fh40000gn/T/pip-install-ulu9kzcd/dukpy_a71c397eaff2464696eb4d0fdd56caab/setup.py'"'"';f
= getattr(tokenize, '"'"'open'"'"', open)(__file__) if
os.path.exists(__file__) else io.StringIO('"'"'from setuptools import
setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"',
'"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))'
install --record
/private/var/folders/9s/3hlz7cm91n7975n552pg6fh40000gn/T/pip-record-g3mqsqim/install-record.txt
--single-version-externally-managed --compile --install-headers
/Library/Frameworks/Python.framework/Versions/3.8/include/python3.8/dukpy
Check the logs for full command output.

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
04/26/21,
02:09:47 PM

ErikBjare, 2021-04-26:
Email replies makes your comment impossible to format with Markdown and therefore excessively long. So here it is with Markdown:

```
WARNING: Value for scheme.headers does not match. Please report this to <
https://github.com/pypa/pip/issues/9617>

distutils:
/Library/Frameworks/Python.framework/Versions/3.8/include/python3.8/UNKNOWN

sysconfig:
/Library/Frameworks/Python.framework/Versions/3.8/include/python3.8

WARNING: Additional context:

user = False

home = None

root = None

prefix = None

Collecting dukpy

  Using cached dukpy-0.2.3.tar.gz (1.9 MB)

Using legacy 'setup.py install' for dukpy, since package 'wheel' is not
installed.

Installing collected packages: dukpy

    Running setup.py install for dukpy ... error

    ERROR: Command errored out with exit status 1:

     command:
/Library/Frameworks/Python.framework/Versions/3.8/bin/python3.8 -u -c
'import io, os, sys, setuptools, tokenize; sys.argv[0] =
'"'"'/private/var/folders/9s/3hlz7cm91n7975n552pg6fh40000gn/T/pip-install-ulu9kzcd/dukpy_a71c397eaff2464696eb4d0fdd56caab/setup.py'"'"';
__file__='"'"'/private/var/folders/9s/3hlz7cm91n7975n552pg6fh40000gn/T/pip-install-ulu9kzcd/dukpy_a71c397eaff2464696eb4d0fdd56caab/setup.py'"'"';f
= getattr(tokenize, '"'"'open'"'"', open)(__file__) if
os.path.exists(__file__) else io.StringIO('"'"'from setuptools import
setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"',
'"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))'
install --record
/private/var/folders/9s/3hlz7cm91n7975n552pg6fh40000gn/T/pip-record-g3mqsqim/install-record.txt
--single-version-externally-managed --compile --install-headers
/Library/Frameworks/Python.framework/Versions/3.8/include/python3.8/dukpy

         cwd:
/private/var/folders/9s/3hlz7cm91n7975n552pg6fh40000gn/T/pip-install-ulu9kzcd/dukpy_a71c397eaff2464696eb4d0fdd56caab/

    Complete output (145 lines):

    running install

    running build

    running build_py

    creating build

    creating build/lib.macosx-10.9-x86_64-3.8

    creating build/lib.macosx-10.9-x86_64-3.8/dukpy

    copying dukpy/run.py -> build/lib.macosx-10.9-x86_64-3.8/dukpy

    copying dukpy/__init__.py -> build/lib.macosx-10.9-x86_64-3.8/dukpy

[... 391 lines cut ...]
/Library/Developer/CommandLineTools/usr/bin/xcrun

    error: command 'gcc' failed with exit status 1

----------------------------------------

ERROR: Command errored out with exit status 1:
/Library/Frameworks/Python.framework/Versions/3.8/bin/python3.8 -u -c
'import io, os, sys, setuptools, tokenize; sys.argv[0] =
'"'"'/private/var/folders/9s/3hlz7cm91n7975n552pg6fh40000gn/T/pip-install-ulu9kzcd/dukpy_a71c397eaff2464696eb4d0fdd56caab/setup.py'"'"';
__file__='"'"'/private/var/folders/9s/3hlz7cm91n7975n552pg6fh40000gn/T/pip-install-ulu9kzcd/dukpy_a71c397eaff2464696eb4d0fdd56caab/setup.py'"'"';f
= getattr(tokenize, '"'"'open'"'"', open)(__file__) if
os.path.exists(__file__) else io.StringIO('"'"'from setuptools import
setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"',
'"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))'
install --record
/private/var/folders/9s/3hlz7cm91n7975n552pg6fh40000gn/T/pip-record-g3mqsqim/install-record.txt
--single-version-externally-managed --compile --install-headers
/Library/Frameworks/Python.framework/Versions/3.8/include/python3.8/dukpy
Check the logs for full command output.
```

As the logs indicate, you are missing xcrun (and probably more) from Xcode Developer Tools. This is a common dependency for building almost any software on macOS, and fails for this particular package because it doesn't provide binary wheels for macOS (as indicated in the issue I linked above). 

Installing Xcode Developer Tools (or whatever it is called) should fix it.

ErikBjare, 2021-04-27:
Closing since this is certainly due to missing Xcode dev tools. It'd be nice if they weren't needed, but it's up to the dukpy devs to fix it by publishing binary wheels.

closed 2021-04-27 by ErikBjare

pikawika, 2021-11-03:
I had a similar issue, mine was due to missing exports. See [here](https://github.com/astropy/halotools/issues/813#issuecomment-552165844). Running the install procedure of eeg-notebooks once more after doing these 2 exports worked.

WeiShaoD, 2022-11-14:
Hello, everyone, I'm wondering if there is any update for this issue? since I also met the same problem. my MacBook is Pro M1 with system 13.0.

Thanks

ErikBjare, 2022-11-24:
@WeiShaoD Did you install Xcode developer tools as suggested?