NeuroTechX/EEG-ExPy #97: Add signal quality checks before recording starts

oreHGA, 2021-05-06:
## 📝 Provide a description of the new feature

_What is the expected behavior of the proposed feature?_

This will be based off @ErikBjare 's take on the problem - https://erik.bjareholt.com/thesis/Signal.html

- We will add a new flag to command line to check signal quality : eegnb runexp [flag for quality check]

Results will be placed on the command line prompt with a suggestion on changes to make to improve quality.

_What is the scenario this would be used?_

Before the recording gets started, the quality check happens to make the user aware of how valuable their recordings will be for analysis moving forward.

Initially targeting muse devices, but we aim to generalize to different device types.

---

If you'd like to see this feature implemented, add a 👍 reaction to this post.

JohnGriffiths, 2021-05-20:
This function should go in the cli `utils.py` file:

https://github.com/JohnGriffiths/eeg-notebooks/blob/master/eegnb/cli/utils.py

Made a minimal start on this for muselsl. Pushing shortly.

JohnGriffiths, 2021-05-20:
OR, maybe we should make use of our device abstraction and put some part of this as a method in the `EEG` class

https://github.com/NeuroTechX/eeg-notebooks/blob/master/eegnb/devices/eeg.py

JohnGriffiths, 2021-05-20:
Make a 10s temp recording file for sigqual eval with EEG device and muse headset

```python
import os
import time
from eegnb.devices.eeg import EEG
f = os.path.abspath('sigqual_tempfile.csv')
this_eeg = EEG(device='muse2016')
this_eeg.start(f, duration=10)
time.sleep(3) # get error if no markers?
this_eeg.push_sample([99], timestamp=time.time())
```

JohnGriffiths, 2021-05-20:
Code for pulling samples used in the muselsl visualizer

https://github.com/alexandrebarachant/muse-lsl/blob/master/muselsl/viewer_v2.py

```python
from pylsl import StreamInlet, resolve_byprop
from muselsl.constants  import LSL_EEG_CHUNK, LSL_SCAN_TIMEOUT

streams = resolve_byprop('type', 'EEG', timeout=LSL_SCAN_TIMEOUT)
inlet = StreamInlet(streams[0], max_chunklen=LSL_EEG_CHUNK)
samples, timestamps = inlet.pull_chunk(timeout=0.0, max_samples=100)

import numpy as np
arr = np.array(samples)
np.std(arr,axis=0)

```

ErikBjare, 2021-05-20:
Brainflow filtering: https://brainflow.readthedocs.io/en/stable/Examples.html#python-signal-filtering

MNE filtering: https://mne.tools/stable/generated/mne.filter.filter_data.html

My filtering: https://github.com/ErikBjare/thesis/blob/b25661aaffa5ab957b464fe8f46fdbfe81b51eb4/src/eegclassify/clean.py#L26-L45

closed 2021-07-18 by JohnGriffiths