NeuroTechX/EEG-ExPy #102: Add a CLI experiment run to the pre-check in tests

oreHGA, 2021-05-20:
## 📝 Provide a description of the new feature

_What is the expected behavior of the proposed feature?  What is the scenario this would be used?_

Add a mock run of the cli to allow for end-to-end test of an eeg-notebook experiment. Code can live in the `/test` folder.

Eventually, this test methods should run successfully before completion.

* you can initialize a mock device with brainflow
---

If you'd like to see this feature implemented, add a 👍 reaction to this post.

ErikBjare, 2021-05-20:
Perhaps https://github.com/NeuroTechX/eeg-notebooks/pull/88 can be continued.

ErikBjare, 2021-05-20:
I fixed and merged #88, so now we have a framework for adding more tests at least.