NeuroTechX/EEG-ExPy #104: how to visualize

andraderenew, 2021-05-21:
Dear community:
My question goes to how to run the code in eegnb examples with this command "python *viz*". When I run that I have to change certain things inside the code of course. So I change 'eegnb_examples' for 'local' and 'muse2016' for 'muse2'. Still that doesn't make the trick. What shall I do further on?

JohnGriffiths, 2021-05-21:
Hi there. 

The problem seems to be that the script is not finding any recording files. 

Are you wanting to run this on the eegnb example dataset or on your own recordings? 

In either case, check that the file path being generated at the start of the script is indeed pointing to a folder with .csv files in on your Machine.

andraderenew, 2021-05-21:
I want to run this on a file being recorded with eegnb although I don't know how to check the path is the correct one. Shall I print?

andraderenew, 2021-05-22:
the issue for me here is that I can check tht output of these two lines by printing 

eegnb_data_path = os.path.join(os.path.expanduser('~/'),'.eegnb', 'data')
n170_data_path = os.path.join(eegnb_data_path, 'visual-N170', 'local')

but not of this line

raw = load_data(subject,session,
                experiment='visual-N170', site='eegnb_examples', device_name='muse2',
                data_dir = eegnb_data_path)

this is something I don't know how it reads inside. 

I attach the code I changed

eegnb_data_path = os.path.join(os.path.expanduser('~/'),'.eegnb', 'data')
n170_data_path = os.path.join(eegnb_data_path, 'visual-N170', 'local')

# If dataset hasn't been downloaded yet, download it
if not os.path.isdir(n170_data_path):
    fetch_dataset(data_dir=eegnb_data_path, experiment='visual-N170', site='eegnb_examples');

subject = 1
session = 1
raw = load_data(subject,session,
                experiment='visual-N170', site='eegnb_examples', device_name='muse2',
                data_dir = eegnb_data_path)

I don't know if I have to change 'eegnb_examples' to 'local'

andraderenew, 2021-05-22:
I have tried with python console and I used local as well and nothing seems to work. It says it expects 6 arguments and saw 6. 

raw = load_data(subject, session,  experiment='visual-N170',site='local',device_name='muse2', data_dir='~/.eegnb/data/')

oreHGA, 2021-06-18:
> Error tokenizing data. C error: Expected 6 fields in line 14, saw 7

Hi @andraderenew, reading through the parser error, I think this may be from the recorded .csv file... can you check the content of the .csv file on like 14 and see how many entries you have?

ErikBjare, 2021-06-18:
@oreHGA The reason is due to an unmarked Markers column, see: https://github.com/NeuroTechX/eeg-notebooks/issues/108

andraderenew, 2021-06-20:
I added 14 zeros(which I figured is what could fix this correct me if I'm wrong) and now I have this error

Traceback (most recent call last):
  File "01r__n170_viz.py", line 61, in <module>
    raw = load_data(subject,session,
  File "/Users/andraderenew/eeg-notebooks/eegnb/analysis/utils.py", line 157, in load_data
    return load_csv_as_raw(
  File "/Users/andraderenew/eeg-notebooks/eegnb/analysis/utils.py", line 84, in load_csv_as_raw
    raws = concatenate_raws(raw, verbose=verbose)
  File "<decorator-gen-176>", line 22, in concatenate_raws
  File "/Users/andraderenew/opt/anaconda3/lib/python3.8/site-packages/mne/io/base.py", line 2213, in concatenate_raws
    raws[0].append(raws[1:], preload)
IndexError: list index out of range

ErikBjare, 2021-06-20:
> File "", line 22, in concatenate_raws

Did you also add a header "Marker1" or such?

andraderenew, 2021-06-25:
sorry I didn't know, so now I did add it but I still have this error. 

(eeg-notebooks) andraderenew@Renes-MacBook-Air visual_n170 % python *viz*
Traceback (most recent call last):
  File "01r__n170_viz.py", line 63, in <module>
    data_dir = eegnb_data_path)
  File "/Users/andraderenew/eeg-notebooks/eegnb/analysis/utils.py", line 163, in load_data
    verbose=verbose,
  File "/Users/andraderenew/eeg-notebooks/eegnb/analysis/utils.py", line 84, in load_csv_as_raw
    raws = concatenate_raws(raw, verbose=verbose)
  File "<decorator-gen-191>", line 22, in concatenate_raws
  File "/Users/andraderenew/opt/anaconda3/envs/eeg-notebooks/lib/python3.7/site-packages/mne/io/base.py", line 2465, in concatenate_raws
    raws[0].append(raws[1:], preload)
IndexError: list index out of range

oreHGA, 2021-07-02:
Hi @andraderenew:

This error is pointing toward the file path not being specified correctly in the 

- What OS are you using?
- Where does your recorded file live, by default it should be in the `~/.eegnb/` directory. Can you confirm you have recording files present in this directory?

[image]

Be sure that you have folders "subject0001" and "session001" in the data directory... this is what the `subject` and `session` value point to in the code below

- Can you share a screenshot what your recording file looks like? (it helps if we're able to see the first 30 columns at least)

Finally in your code snippet for loading the data, can you ensure it looks like this & give it another run:

```
eegnb_data_path = os.path.join(os.path.expanduser('~/'),'.eegnb', 'data')
n170_data_path = os.path.join(eegnb_data_path, 'visual-N170', 'local')

# If dataset hasn't been downloaded yet, download it
if not os.path.isdir(n170_data_path):
    fetch_dataset(data_dir=eegnb_data_path, experiment='visual-N170', site='eegnb_examples');

subject = 1
session = 1
raw = load_data(subject,session,
                experiment='visual-N170', site='local', device_name='muse2',
                data_dir = eegnb_data_path)
```

andraderenew, 2021-07-24:
> 
> 
> Hi @andraderenew <https://github.com/andraderenew>:
> 
> This error is pointing toward the file path not being specified correctly in the
> 
> What OS are you using?
Mac OS big sur 11.4

> Where does your recorded file live, by default it should be in the ~/.eegnb/ directory. Can you confirm you have recording files present in this directory <https://user-images.githubusercontent.com/13490968/124229045-dd395f00-dac1-11eb-8b00-152189d270c6.png>Here you have the screenshot
> Be sure that you have folders "subject0001" and "session001" in the data directory... this is what the subject and session value point to in the code below
> 
> Can you share a screenshot what your recording file looks like? (it helps if we're able to see the first 30 columns at least)

> Finally in your code snippet for loading the data, can you ensure it looks like this & give it another run:
>
> eegnb_data_path = os.path.join(os.path.expanduser('~/'),'.eegnb', 'data')
> n170_data_path [...]

This is what I get using the code provided!!!

(eeg-notebooks) andraderenew@Renes-MacBook-Air visual_n170 % python *viz*
Traceback (most recent call last):
  File "01r__n170_viz.py", line 63, in <module>
    data_dir = eegnb_data_path)
  File "/Users/andraderenew/eeg-notebooks/eegnb/analysis/utils.py", line 163, in load_data
    verbose=verbose,
  File "/Users/andraderenew/eeg-notebooks/eegnb/analysis/utils.py", line 65, in load_csv_as_raw
    ch_names = [list(data.columns)[i] for i in ch_ind] + ["stim"]
  File "/Users/andraderenew/eeg-notebooks/eegnb/analysis/utils.py", line 65, in <listcomp>
    ch_names = [list(data.columns)[i] for i in ch_ind] + ["stim"]
IndexError: list index out of range

oreHGA, 2023-05-04:
Closing due to inactivity, let us know if you're still running into issues.

closed 2023-05-04 by oreHGA