NeuroTechX/EEG-ExPy #108: Missing Marker Column in Output (Muse2 with BlueMuse)

felixmpichardo, 2021-06-06:
## ℹ Computer information

- Platform OS (e.g Windows, Mac, Linux etc):  Windows 10
- Python Version: 3.7
- Brain Interface Used (e.g Muse, OpenBCI, Notion etc): Muse2 with Bluemuse

## 📝 Provide detailed reproduction steps (if any)

1. Connect to BlueMuse and begin streaming.
2. Ran the demonstration code for running an experiment [as per the documentation ](https://neurotechx.github.io/eeg-notebooks/getting_started/running_experiments.html).  I tried 30s, 120s, visual_n170, and visual_p300.
3. Additionally, I ran [a snippet of code for "signal quality"](https://github.com/NeuroTechX/eeg-notebooks/issues/97#issuecomment-845141349), and I send a couple of test markers.

### ✔️ Expected result

The fetched data looked like this:

[image]

I expected all the columns except the "Right AUX" column.

### ❌ Actual result
(1) The result of running the experiment:
There is no marker column at all:
[image]

(2) Signal snippet: 
Initially, there was no marker column, as above. This morning, it began to show up, but it has no header or data for the first few lines, which causes import errors. The other markers I send in the snippet do show up.

[image]

ErikBjare, 2021-06-18:
Thinking about this again, I think I may be the one to blame. Regression probably happened in my PR to muse-lsl: https://github.com/alexandrebarachant/muse-lsl/pull/139

Maybe we should open an issue there instead? Or just figure out how to fix it (shouldn't be too hard) and submit a PR straight away.

Maari13, 2021-06-18:
I am a complete novice as far as EEG-Notebooks or the topic of streaming goes. Just wanted to say, that I had the same issue (#111) in both cases: when I used BlueMuse for streaming and when I used Petal Metrics. Petal Metrics gives the option to choose either LSL or OSC as the stream type. I previously used LSL, but just now tried out the OSC option, however the problem persists - at the beginning the event markers are not recorded - the first one that gets recorded is always on data row 13 (row 14 in the file).

Maari13, 2021-06-21:
In order to do some further testing, I also experimented with Visual N170, Visual SSVEP and Auditory Oddball (orig) - in all the cases the title of the "Marker" column was missing from the experimental data file, as well as the first 12 event markers. Hence, the issue is not specific to the Visual P300 experiment.