NeuroTechX/EEG-ExPy #111: Visual P300 event markers not recorded

Maari13, 2021-06-17:
## ℹ Computer information

- Platform OS: Windows 10
- Python Version: Python 3.7.10
- Brain Interface Used: Muse 2

## 📝 Provide detailed reproduction steps (if any)

1. switch on your Muse device
2. start streaming using BlueMuse or Petal Metrics (I tried with both and had the same issues)
3. run the experiment with the command eegnb runexp -ip, and input the necessary parameters
4. complete the experiment

### ✔️ Expected result

The title of the last column of the output/experimental data file is "Marker" and below it starting from the next row are zero markers denoting that no picture is presented at this time

### ❌ Actual result
The title of the last column is missing, and zeros start from data row 13 (row 14 of the file)

## 📷 Screenshots (this experimental was done using BlueMuse for streaming)
[image]

ErikBjare, 2021-06-18:
I think this is a duplicate of #108

Maari13, 2021-06-18:
The column is missing values initially: starting from data row 13 (or row 14 of the file) until the end of the file the values are present. I was wondering whether I can trust that the values that are present are aligned with the correct timestamp, or the fact that the markers start from row 13/14 means that they are shifted in time - I mean the marker on row 14 should really be on row 2?

ErikBjare, 2021-06-18:
I'm fairly confident that they are aligned, but @JohnGriffiths might have a more authoritative answer.

JohnGriffiths, 2021-06-18:
I agree. The lsl process writes to file what is received on the the current stream at each time point. I'm not aware of any way it would buffer or otherwise delay the stream output. Especially from the markers stream, which is running locally (as opposed to the EEG stream, which operates on Bluetooth data packets that are known to have a minimum and a variable lag).

rschmaelzle, 2022-02-06:
Hi, I can confirm this issue. I was able to track it down - in part - to the muselsl-record function, where there is an if-condition that has a "save every 5s"-comment. Messing with this "catches" more markers (although it doesn't yet cure my problem).  Hope this helps. I will keep digging myself into it a bit more.

tmorshed, 2024-04-04:
Closing to concatenate in #108.

closed 2024-04-04 by tmorshed