NeuroTechX/EEG-ExPy #112: Number of events 0

Maari13, 2021-06-28:
## ℹ Computer information

- Platform OS: Windows 10
- Python Version: Python 3.7.10
- Brain Interface Used: Muse 2

## 📝 Provide detailed reproduction steps (if any)

1. switch on Muse 2 and start streaming with Petal Metrics
2. run experiment P300 with the command eegnb runexp -ip, and input the necessary parameters; complete experiment
3. fill in the missing values in the experimental data csv file(see #108, #111; without filling in the values you get a tokenizing error)
4. analyze your data as explained in https://neurotechx.github.io/eeg-notebooks/auto_examples/visual_p300/01r__p300_viz.html using Jupyter Notebooks up to and including the epoching step (# Create an array containing the timestamps and type of each stimulus)

### ✔️ Expected result

Based on the marker column a number of events is detected, including both target and non-target events. The output resulting from running this cell shows the corresponding results 

### ❌ Actual result

The output shows that 23 events were found, however, in the table below the number of events is 0, and there are 0 occurrences of both target and non-target events

## 📷 Screenshots

[image]

I believe that this issue is connected to #108 / #111 . If I use the example data (subject 1, session 1) that is used in https://neurotechx.github.io/eeg-notebooks/auto_examples/visual_p300/01r__p300_viz.html, target and non-target events are found. Although, there is a discrepancy in events found (1161) and the number of events in the table below (1143: 959 non-target and 184 target events).

[image]

By visual inspection I do not see any differences between my experimental data file and that of example data (except that I only recorded a single run of 120s, while example data had 6 runs all of longer duration).

I add my experimental data file (had to change the extension from csv to txt, since otherwise I received a file type not supported error when trying to upload it here).
[recording_2021-06-28-14.34.59.txt](https://github.com/NeuroTechX/eeg-notebooks/files/6727878/recording_2021-06-28-14.34.59.txt)

oreHGA, 2021-07-02:
> By visual inspection I do not see any differences between my experimental data file and that of example data (except that I only recorded a single run of 120s, while example data had 6 runs all of longer duration).

@Maari13, ran through your file locally and this seems to be the reason why you're not getting events returned.

In the notebook, you'd notice that this snippet exists:

```# Create an array containing the timestamps and type of each stimulus (i.e. face or house)
events = find_events(raw)
event_id = {'Non-Target': 1, 'Target': 2}
epochs = Epochs(raw, events=events, event_id=event_id,
                tmin=-0.1, tmax=0.8, baseline=None,                                                                                  
                reject={'eeg': 100e-6}, preload=True,                                                                                  
                verbose=False, picks=[0,1,2,3])

print('sample drop %: ', (1 - len(epochs.events)/len(events)) * 100)

epochs
```

The `reject` field determines the threshold for rejecting epochs based on peak-to-peak signal amplitude. [See MNE docs here](https://mne.tools/stable/generated/mne.Epochs.html#mne.Epochs) 

Notice below how changing the `reject` value to a much higher number keeps more events but potentially including more noisy results. 

[image]

My recommendation here would be 
- doing a much longer recording maybe 5mins
- ensure that electrodes are well placed to ensure good quality recording
- tweaking reject value iteratively

JohnGriffiths, 2021-07-02:
Thank you @oreHGA - indeed this is not an uncommon issue. As Ore said, you can try tweaking the threshold iteratively; e.g. up/down a few percent. You will probably see as you do that more epochs are kept the more lenient you go, and concurrently that the average ERP trace gets noisier ('wigglier'), because more noisy data points are being included. 

As Ore said, the fundamental issue here is a noisy recording set up. 

Try 

using the muselsl signal viewer to confirm you have good signal before running the experiment 

`muselsl view -v 2` 

(you may have to also `pip install vispy` first )

And make sure the numbers on the right are <20. Then kill the viewer and run the eegnb expt. 

We have nearly finished a simple signal quality check function for eeg notebooks that will help with this part of the setup process. Watch this space (or watch the PRs if you're very interested). 

Let us know if this helps!

Maari13, 2021-07-02:
Thank you so much for the explanation and advice. It did not occur to me that an automatic quality check was applied to the epoch data. I was going to do longer recordings, but first wanted to make sure I got things to work and then was puzzled ... I will definitely need to check the contact of the electrodes with the scalp and preview the signal before acquiring data next time. And then tweek the reject value

oreHGA, 2021-11-26:
Hi @Maari13 , hope you were able to get more sample in after adjusting the "reject" threshold. I'll go ahead and close this now but feel free to reply to the thread if you still have any issues. Cheers

closed 2021-11-26 by oreHGA