NeuroTechX/EEG-ExPy #116: Make data zipper function

JohnGriffiths, 2021-07-18:


JohnGriffiths, 2021-07-18:
## 1. Make core function
--------------------------

Make python function in `datasets.py` that zips all files in the `ntcs` data folder

Full file path would e.g. be

`~/.eegnb/data/visual-N170/local_ntcs/museS/subject0999/session001/recording_2020-10-19-00.45.04.csv`

We would want to zip up everything below here

`~/.eegnb/data/visual-N170/local_ntcs`

(question mark around parametrization of experiment name here)

Function should zip the folder and create a `.zip` file in a obvious location (e.g. Desktop), and say that clearly in command line output. 

## 2. Make command line program
----------------------------------------

Follow examples of e.g. `checksigqual` in `eegnb.cli.__main__.py`

## 3. Add usage description + explanation to documentation
----------------------------------------

( Note: this is specific to NTCS, not 'generic documentation' )

HVjay, 2021-07-28:
To do next (@HVjay , @AyrusCode ):

Add a bit more feedback at the terminal
( e.g. "running data zipper... your data folder is X...looking for folders in X... X folders found...Making a zip file...Your zip file is at X location...It contains the following directories")
(just some initial thoughts. Don't need to go overboard, but the feedback will be helpful for people to understand what's going on, and also helpful for us in debugging )
Add description of usage to the docs
Add option to specify site
( e.g. eegnb runzip -ex visual-N170 -s local-ntcs-2 )

closed 2021-09-16 by HVjay