NeuroTechX/EEG-ExPy #119: while installing on ubuntu gives wxpython error

kaushikCanada, 2021-07-18:
## ℹ Computer information

- Platform OS (e.g Windows, Mac, Linux etc): 
- Python Version:
- Brain Interface Used (e.g Muse, OpenBCI, Notion etc):

## 📝 Provide detailed reproduction steps (if any)

1. …
2. …
3. …

### ✔️ Expected result

_What is the expected result of the above steps?_

### ❌ Actual result

_What is the actual result of the above steps?_

## 📷 Screenshots

_Are there any useful screenshots? WinKey+Shift+S (Windows) / Shift+Command+4 (MacOS) and then just paste them directly into the form_

JohnGriffiths, 2021-07-18:
Hi Kaushik. 

Thanks for this. 

Just to confirm: this was the result you got when running 

`pip install -e .` 

in the eeg-notebooks folder?

ta.

kaushikCanada, 2021-07-18:
Yea that is the one.

ErikBjare, 2021-07-19:
Try passing `--find-links=https://extras.wxpython.org/wxPython4/extras/linux/gtk3/ubuntu-20.04/` to the pip install command, or install a compiler toolchain (as the error complains about not finding one).

kaushikCanada, 2021-07-19:
Wbat is yhe compiler toolchain? I thought u just do the pip install

ErikBjare, 2021-07-19:
@kaushikCanada See for example: https://stackoverflow.com/a/19816655/965332

JohnGriffiths, 2021-11-26:
The `wxpython' installation issue is solved in the latest documentation ( basically, use conda rather than pip for this ). 

c.f. here

https://github.com/NeuroTechX/eeg-notebooks/issues/138

and the installation instructions ( `wxpython` is in the conda create line )

https://neurotechx.github.io/eeg-notebooks/getting_started/installation.html#installing-the-python-library

closed 2021-11-26 by JohnGriffiths