NeuroTechX/EEG-ExPy #120: OpenGL error when importing n170 task from installation/test page instructions

sjburwell, 2021-07-19:
## ℹ Computer information

- Platform OS (e.g Windows, Mac, Linux etc): MacOS
- Python Version: 3.7.10 (default, Feb 26 2021, 10:16:00)
- Brain Interface Used (e.g Muse, OpenBCI, Notion etc): Notion, Muse

## 📝 Provide detailed reproduction steps (if any)

1. Followed the [installation and test instructions](https://neurotechx.github.io/eeg-notebooks/getting_started/installation.html) verbatim, except, where "eeg-notebooks" I created and activated a separate environment "eeg-notebooks-20210719"
2. Executed the line: `from eegnb.experiments.visual_n170 import load_data`

### ✔️ Expected result

Expected to import the n170 task as in the installation page.

### ❌ Actual result

Returned an OpenGL error: 
`ImportError: ('Unable to load OpenGL library', 'dlopen(OpenGL, 10): image not found', 'OpenGL', None)`

Is there something incorrect about me changing the environment name from "eeg-notebooks" to "eeg-notebooks-20210719" that is problematic? I have another, older, conda environment named "eeg-notebooks" that does not cause errors, but I do not want to remove or change that environment for fear of losing all functionality.

Full error printed below:
```
--------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
~/anaconda3/envs/eeg-notebooks-20210719/lib/python3.7/site-packages/OpenGL/platform/darwin.py in GL(self)
     37                 'OpenGL',
---> 38                 mode=ctypes.RTLD_GLOBAL
     39             ) 

~/anaconda3/envs/eeg-notebooks-20210719/lib/python3.7/site-packages/OpenGL/platform/ctypesloader.py in loadLibrary(dllType, name, mode)
     35     else:
---> 36         return _loadLibraryWindows(dllType, name, mode)
     37 

~/anaconda3/envs/eeg-notebooks-20210719/lib/python3.7/site-packages/OpenGL/platform/ctypesloader.py in _loadLibraryWindows(dllType, name, mode)
     88     try:
---> 89         return dllType( name, mode )
     90     except Exception as err:

~/anaconda3/envs/eeg-notebooks-20210719/lib/python3.7/ctypes/__init__.py in __init__(self, name, mode, handle, use_errno, use_last_error)
    363         if handle is None:
--> 364             self._handle = _dlopen(self._name, mode)
    365         else:

OSError: ('dlopen(OpenGL, 10): image not found', 'OpenGL', None)

During handling of the above exception, another exception occurred:

ImportError                               Traceback (most recent call last)
<ipython-input-4-536bb5b50347> in <module>
      2 from eegnb import generate_save_fn
      3 from eegnb.devices.eeg import EEG
----> 4 from eegnb.experiments.visual_n170 import n170
      5 #from eegnb.analysis.utils import load_data

~/Documents/eeg-notebooks-20210719/eegnb/experiments/visual_n170/n170.py in <module>
      7 import numpy as np
      8 from pandas import DataFrame
----> 9 from psychopy import visual, core, event
     10 
     11 from eegnb import generate_save_fn

~/anaconda3/envs/eeg-notebooks-20210719/lib/python3.7/site-packages/psychopy/visual/__init__.py in <module>
     28 from .button import ButtonStim
     29 from .brush import Brush
---> 30 from .textbox2.textbox2 import TextBox2
     31 # window, should always be loaded first
     32 from .window import Window, getMsPerFrame, openWindows

~/anaconda3/envs/eeg-notebooks-20210719/lib/python3.7/site-packages/psychopy/visual/textbox2/__init__.py in <module>
      2 # -*- coding: utf-8 -*-
      3 
----> 4 from .textbox2 import TextBox2, allFonts

~/anaconda3/envs/eeg-notebooks-20210719/lib/python3.7/site-packages/psychopy/visual/textbox2/textbox2.py in <module>
     18 """
     19 import numpy as np
---> 20 import OpenGL.GL as gl
     21 
     22 from ..basevisual import BaseVisualStim, ColorMixin, ContainerMixin

~/anaconda3/envs/eeg-notebooks-20210719/lib/python3.7/site-packages/OpenGL/GL/__init__.py in <module>
      1 """OpenGL.GL, the core GL library and extensions to it"""
      2 # early import of our modules to prevent import loops...
----> 3 from OpenGL import error as _error
      4 from OpenGL.GL.VERSION.GL_1_1 import *
      5 from OpenGL.GL.pointers import *

~/anaconda3/envs/eeg-notebooks-20210719/lib/python3.7/site-packages/OpenGL/error.py in <module>
     10 import logging
     11 _log = logging.getLogger( 'OpenGL.error' )
---> 12 from OpenGL import platform, _configflags
     13 from ctypes import ArgumentError
     14 __all__ = (

~/anaconda3/envs/eeg-notebooks-20210719/lib/python3.7/site-packages/OpenGL/platform/__init__.py in <module>
     34     return plugin
     35 
---> 36 _load()
     37 
     38 def types(resultType,*argTypes):

~/anaconda3/envs/eeg-notebooks-20210719/lib/python3.7/site-packages/OpenGL/platform/__init__.py in _load()
     31 
     32     # install into the platform module's namespace now
---> 33     plugin.install(globals())
     34     return plugin
     35 

~/anaconda3/envs/eeg-notebooks-20210719/lib/python3.7/site-packages/OpenGL/platform/baseplatform.py in install(self, namespace)
     95         """Install this platform instance into the platform module"""
     96         for name in self.EXPORTED_NAMES:
---> 97             namespace[ name ] = getattr(self,name,None)
     98         namespace['PLATFORM'] = self
     99         return self

~/anaconda3/envs/eeg-notebooks-20210719/lib/python3.7/site-packages/OpenGL/platform/baseplatform.py in __get__(self, obj, cls)
     13         self.fget = function
     14     def __get__( self, obj, cls ):
---> 15         value = self.fget( obj )
     16         setattr( obj, self.fget.__name__, value)
     17         return value

~/anaconda3/envs/eeg-notebooks-20210719/lib/python3.7/site-packages/OpenGL/platform/darwin.py in GetCurrentContext(self)
     60     @baseplatform.lazy_property
     61     def GetCurrentContext( self ):
---> 62         return self.CGL.CGLGetCurrentContext
     63 
     64     def getGLUTFontPointer( self, constant ):

~/anaconda3/envs/eeg-notebooks-20210719/lib/python3.7/site-packages/OpenGL/platform/baseplatform.py in __get__(self, obj, cls)
     13         self.fget = function
     14     def __get__( self, obj, cls ):
---> 15         value = self.fget( obj )
     16         setattr( obj, self.fget.__name__, value)
     17         return value

~/anaconda3/envs/eeg-notebooks-20210719/lib/python3.7/site-packages/OpenGL/platform/darwin.py in CGL(self)
     43     def GLU(self): return self.GL
     44     @baseplatform.lazy_property
---> 45     def CGL(self): return self.GL
     46 
     47     @baseplatform.lazy_property

~/anaconda3/envs/eeg-notebooks-20210719/lib/python3.7/site-packages/OpenGL/platform/baseplatform.py in __get__(self, obj, cls)
     13         self.fget = function
     14     def __get__( self, obj, cls ):
---> 15         value = self.fget( obj )
     16         setattr( obj, self.fget.__name__, value)
     17         return value

~/anaconda3/envs/eeg-notebooks-20210719/lib/python3.7/site-packages/OpenGL/platform/darwin.py in GL(self)
     39             ) 
     40         except OSError as err:
---> 41             raise ImportError("Unable to load OpenGL library", *err.args)
     42     @baseplatform.lazy_property
     43     def GLU(self): return self.GL

ImportError: ('Unable to load OpenGL library', 'dlopen(OpenGL, 10): image not found', 'OpenGL', None)
```

sjburwell, 2021-07-21:
I think I have resolved the issue by updating psychopy to version 2021.2.0, per the PsychoPy installation instructions! Not sure if there are any caveats to running a more current version of psychopy with these notebooks, but posting here in case anyone else runs into the same error and/or the "official" psychopy version could be updated. 
[image]

closed 2021-07-21 by sjburwell

JohnGriffiths, 2021-07-21:
@sjburwell - out of interest, do you get an error with the following:

- Create a fresh conda environment
- install / update pip
- pip install psychopy
- `import psychopy`
- `import eegnb`
- `from eegnb.experiments.visual_n170 import load_data`

In other words, is this an issue with `pip` psychopy installation, on your OS (Mac)

sjburwell, 2021-07-29:
Apologies for the delay - just got around to trying this in the order you suggest. Same issues - unable to load OpenGL. With pip 21.2.1, followed by a psychopy installation. 

(Ps. I realized in the above code I reference `from eegnb.experiments.visual_n170 import load_data` being the problematic line, rather it's the line `from eegnb.experiments.visual_n170 import n170`. Same issue, nonetheless.)