NeuroTechX/EEG-ExPy #127: When trying to use brainflow==4.4.0, "stim" channel all 0s in saved .csv

sjburwell, 2021-07-30:
## ℹ Computer information

- Platform OS (e.g Windows, Mac, Linux etc): MacOS
- Python Version: 3.7
- Brain Interface Used (e.g Muse, OpenBCI, Notion etc): Notion 2, Crown

## 📝 Provide detailed reproduction steps (if any)

I am trying to record data from the Neurosity Crown, which was included in `brainflow==4.4.0`. I have updated `brainflow` in my local `eeg-notebooks` environment, as well as added "crown" below "notion2" in the relevant `eegnb/devices/*` and `eegnb/cli/*` files. 

### ✔️ Expected result

I would like to record data from Neurosity Crown alongside event markers, I expected that the saved .csv file would contain both EEG data and event marker data.

### ❌ Actual result

I am able to record Crown data and the EEG channel data are successfully saved to the .csv file, but **the "stim" column of the .csv is all 0s.** The same experiment ran fine and saved both EEG and "stim" data .csv without updating `brainflow==3.7.2`; when I update to `brainflow==4.4.0` the "stim" column is all 0s for Crown AND Notion 2. As such, there is an incompatibility between the most recent version(s) of `brainflow` and `eeg-notebooks`. 

**My question is:** 
Does this issue look familiar or might you know of a simple fix for it? And/or, are there plans in the works to make `brainflow==4.4.0` compatible with `eeg-notebooks`?

Thanks!

JadinTredup, 2021-08-12:
@sjburwell my bad for not seeing this earlier going to try to solve this today. Will update you when I get make some progress.

sjburwell, 2021-08-12:
Thanks @JadinTredup - please let me know how I might help with testing, etc. I was able to integrate Crown into my local `eeg-notebooks` repo, but the events (stimulus timing, etc.) aren't coming through, but I imaging someone more familiar with `eeg-notebooks` might have an idea where the problem may be coming from.

JadinTredup, 2021-08-12:
@sjburwell I believe I have fixed the problem in this PR. I tested it with my Notion 2 but if you could give it a test with your Crown before I merge it, that would be greatly appreciated!

sjburwell, 2021-08-13:
Just posted to the PR, but I just tested on the Crown and worked well for me! Thanks @JadinTredup for handling this!

PR #128 (2021-08-13): Update to support newest brainflow version (4.5.0)
I added support for the newest Brainflow version which includes support for the Neurosity Crown. I also fixed an issue where the newest brainflow would cause stim markers to not save in the CSV file.

Resolves #127
closed 2021-08-13 by sjburwell