NeuroTechX/EEG-ExPy #132: Issue initializing Muse2 on MacOs

pjkohler, 2021-09-15:
## ℹ Computer information

- Platform OS (e.g Windows, Mac, Linux etc): macOS 10.15
- Python Version: 3.7
- Brain Interface Used (e.g Muse, OpenBCI, Notion etc): Muse with BLED112 dongle

## 📝 Provide detailed reproduction steps (if any)

1. Install eeg-notebooks following the instructions here: https://neurotechx.github.io/eeg-notebooks/getting_started/installation.html
2. Attempt to open EEG stream, using this code (adapted from [this](https://neurotechx.github.io/eeg-notebooks/auto_examples/visual_n170/00x__n170_run_experiment.html)): 
```
import os
import sys
sys.path.insert(1, '/Users/kohler/code/git/eeg-notebooks')
from eegnb import generate_save_fn
from eegnb.devices.eeg import EEG
from eegnb.experiments.visual_n170 import n170
eeg_device = EEG(device=board_name)

board_name = "muse2"
experiment = "visual_n170"
subject_id = 0
session_nb = 0
record_duration = 120

eeg_device = EEG(device=board_name)
```

### ✔️ Expected result

EEG device initialization

### ❌ Actual result

muse_get_recent throws an error, message "Couldn't find any stream, is your device connected?"

Note, connecting directly with muselsl like this:
```
from muselsl import stream, view, list_muses

muses = list_muses()
stream(muses[0]['address'])
```
works fine. 

## 📷 Screenshots

[image]

pjkohler, 2021-09-15:
Okay, so I think I have to start the stream in a separate terminal window before running the notebook. When I do that, I get a different error. I get the same error whether I use "muse" or "muse2" as the device. 
[image]

oreHGA, 2021-11-26:
Hi @pjkohler - Do you mind giving this a shot again? We have made recent changes that should attempt to fix this

pjkohler, 2021-11-26:
of course! how do I update?

oreHGA, 2021-11-27:
thanks! here are some steps to try

- pulling the latest changes - `git pull origin master`
- (optional) reinstalling dependencies in your environment for the project - `pip install -e .`
- try running the sample here https://neurotechx.github.io/eeg-notebooks/getting_started/installation.html#installing-the-python-library

pjkohler, 2022-03-23:
I get a different error now. See below. Note that I first connected to the Muse device using uvicMuse. Thanks. 
[image]

pjkohler, 2022-03-23:
ignore previous comment (that I deleted), I realize now I also have to start the stream in uvicMuse, before running eeg-notebooks.

pjkohler, 2022-03-23:
okay, now I am able to start the experiment, but then I get the following error after pressing spacebar. Note that I have connected and started streaming using uvicMuse prior to running the eeg-notebooks code.
[image]

pjkohler, 2022-03-23:
I was able to get around that error by adding
`import nest_asyncio
nest_asyncio.apply()`
above. Now I get this error:
[image]

pjkohler, 2022-03-23:
The muse is connected and streaming, I can data from it using LabRecorder software.

JohnGriffiths, 2022-03-23:
Hi Peter.

Have you tried the brainflow option? 

Device name for native Bluetooth with brainflow is 'muse2_bfn'.

JohnGriffiths, 2022-03-23:
Relevant info re: OS compatibility etc here

https://brainflow.readthedocs.io/en/stable/SupportedBoards.html#muse

pjkohler, 2022-03-23:
I haven't. Brainflow would be used to initialize the stream, right?

pjkohler, 2022-03-24:
ah, I misunderstood ... maybe I misunderstood the whole approach you take here. There is no need to initialize the stream ahead of time, is there? I can simply pass "muse2_bfn" as the device name, and then your code will take of the rest, right? It appears so. I was able to get some data, and will continue to debug.

pjkohler, 2022-03-24:
okay, this is pretty awesome. Thanks @JohnGriffiths, @oreHGA, everyone. I will use this in my class on Tuesday, and a lot more in future courses! 

Also, for some reason, not sure why, I do not have to run `import nest_asyncio nest_asyncio.apply()` to get this to work anymore.

JohnGriffiths, 2022-03-24:
Great to hear this is working well for you Peter!

You misunderstanding is very understandable: we are using brainflow differently to how muselsl and BlueMuse are have been used in `eeg-notebooks` previously, namely the stream is initiated within `EEG` device object initialization, without the need of the user to initiate any separate third-party streaming processes, through Python or otherwise. 

BrainFlow is a relatively new library that does a great job of handling data streams across a wide variety of devices. They added muse support just after Xmas, and so we are gradually moving over to that ( mainly `_bfn'  ) as the default / recommended streaming option, whilst continuing to keep the others available. 

You should also check out the CLI if you haven't yet: type 

`eegnb runexp -ip`

and follow the options.

Finally, since you mentioned your class, here is a minimal `eeg-notebooks` script we are using in a current experiment, that may be handy as a refernce

https://github.com/GriffithsLab/muse-aob-tms/blob/main/code/run_aob_task.py

If you configure the 'run with' option in windows/mac to the correct python environment binary, then this can run a complete experiment - including stream initiation and data saving - with a simple double-click of a desktop icon.  In windows I found this works better if I create a separate shortcut that points to the `run_aob_task.py` file. Not sure whether that would be helpful on Mac.

pjkohler, 2022-03-24:
holy smokes, John. I have the N170 experiment working, and that may just have to do for now, since we are coming down to the wire. But I plan to teach this course for many years (Neuroscience Techniques - a sampler of different neuroscience methodologies) so will explore more in the future.

JohnGriffiths, 2022-03-24:
Another little snippet that might be useful

This is a quick way to grab data with the device class without having to run a psychopy experiment or to save data to file

```python
from eegnb.devices.eeg import EEG
myeeg = EEG(device='muse2_bfb') # automatically initiates a brainflow connection
dat = myeeg.get_recent() # returns a time x channel pandas dataframe 
dat.plot() # pandas matplotlib method
```

Unfortunately we don't have a livestreaming figure like you get with `muselsl view -v 2` yet, but we will add one at some point (unless the brainflow guys beat us to it)

In the mean time there is a simple signal quality check

Command lne:
```
eegnb checksigqual -ed muse2_bfn
```

Python:
```python
from eegnb.analysis.utils import check_report
check_report(myeeg)
```

That prints out standard deviations on the command line every 5 secs (like the numbers on the axes in the muselsl view plots ); also a bit like the actual muse app.

pjkohler, 2022-03-24:
okay, so I have eeg-notebooks running on my primary computer, which is running Mac OSX 10.15. One thing I am still confused about is the BLED dongle - so I do not need that at all? 

In parallel, I am struggling with getting eeg-notebooks to run on my secondary computer which has Mac OSX 10.13 - I can share my specific hassles, but any general tips?

pjkohler, 2022-03-25:
okay `from eegnb.analysis.utils import check_report
check_report(myeeg)` is real cool, and should just be run before each block I would think.

oreHGA, 2023-04-27:
Hey @pjkohler curious, is this still an active issue for you?

closed 2023-04-27 by pjkohler