NeuroTechX/EEG-ExPy #150: PsychoPy n170 fails on Windows 10 Pro

sbastiaens, 2022-01-07:
- Windows 10 Pro
- Python Version: 3.7.11
- Brain Interface Used: Crown

When running n170.present (form visual_n170), PsychoPy freezes and python crashes. 
The problem was fixed by removing `show_instructions(duration)`. We think that the error comes from the fact that `mywin = visual.Window([1600, 900], monitor="testMonitor", units="deg", fullscr=True)` is repeated twice and therefore PsychoPy creates two screens which makes it crash.

gbernal, 2022-01-22:
Hi,

I'm seeing similar issues when running the n170 experiment, p300, and so on.  looking adding break points into the code, the code fails when it goes into     `mywin = visual.Window([1600, 900], monitor="testMonitor", units="deg", fullscr=True)`
as mentioned by @sbastiaens I tried commenting out `show_instructions(duration)` but that didn't fix it for me

This is my setup 
Windows 10 
Python Version: 3.9
Brain Interface Used: Cyton

[2022-01-22 18:39:35.565] [board_logger] [info] openning port \\.\COM19
Traceback (most recent call last):
  File "C:\Users\ExpMachine\Documents\GitHub\eeg-notebooks\tests\00x__ssvep_run_experiment.py", line 20, in <module>
    n170.present(duration=record_duration, eeg=eeg_device, save_fn=save_fn)
  File "c:\users\ExpMachine\documents\github\eeg-notebooks\eegnb\experiments\visual_n170\n170.py", line 37, in present
    mywin = visual.Window([1600, 900], monitor="testMonitor", units="deg", fullscr=True)
  File "C:\Users\ExpMachine\miniconda3\envs\eeg-notebooks\lib\site-packages\psychopy\visual\window.py", line 343, in __init__
    self.monitor = monitors.Monitor(monitor, autoLog=autoLog)
  File "C:\Users\ExpMachine\miniconda3\envs\eeg-notebooks\lib\site-packages\psychopy\monitors\calibTools.py", line 127, in __init__
    self._loadAll()
  File "C:\Users\ExpMachine\miniconda3\envs\eeg-notebooks\lib\site-packages\psychopy\monitors\calibTools.py", line 446, in _loadAll
    self.calibs = json_tricks.load(thisFile, ignore_comments=False,
  File "C:\Users\ExpMachine\miniconda3\envs\eeg-notebooks\lib\site-packages\json_tricks\nonp.py", line 266, in load
    return loads(string, preserve_order=preserve_order, ignore_comments=ignore_comments, decompression=decompression,
  File "C:\Users\ExpMachine\miniconda3\envs\eeg-notebooks\lib\site-packages\json_tricks\nonp.py", line 236, in loads
    return json_loads(string, object_pairs_hook=hook, **jsonkwargs)
  File "C:\Users\ExpMachine\miniconda3\envs\eeg-notebooks\lib\json\__init__.py", line 359, in loads
    return cls(**kw).decode(s)
TypeError: __init__() got an unexpected keyword argument 'encoding'
Exception ignored in: <function Window.__del__ at 0x00000224BE696160>
Traceback (most recent call last):
  File "C:\Users\ExpMachine\miniconda3\envs\eeg-notebooks\lib\site-packages\psychopy\visual\window.py", line 582, in __del__
    self.close()
  File "C:\Users\ExpMachine\miniconda3\envs\eeg-notebooks\lib\site-packages\psychopy\visual\window.py", line 2274, in close
    self.backend.close()  # moved here, dereferencing the window prevents
AttributeError: 'NoneType' object has no attribute 'close'

gbernal, 2022-01-23:
I solved this issue by upgrading psychopy to 2021.2.3

ErikBjare, 2022-03-25:
@gbernal Thanks for that, I updated to latest Psychopy in #167 so at least that shouldn't be an issue anymore.

closed 2022-04-25 by oreHGA

rkakrik, 2022-06-30:
Same problem in Arch Linux