NeuroTechX/EEG-ExPy #160: ImportError: Unable to load OpenGL library

porwal97, 2022-03-14:
I am trying to Test if the Python library installed correctly by using the code: 

from eegnb import generate_save_fn
from eegnb.devices.eeg import EEG
from eegnb.experiments.visual_n170 import n170

_I get the following ImportError. Any advice on how to fix this?_ 
**ImportError:** ('Unable to load OpenGL library', "dlopen(OpenGL, 0x000A): tried: '/opt/anaconda3/envs/eeg-notebooks/lib/OpenGL' (no such file), '/opt/anaconda3/envs/eeg-notebooks/lib/OpenGL' (no such file), '/opt/anaconda3/envs/eeg-notebooks/lib/python3.7/lib-dynload/../../OpenGL' (no such file), '/opt/anaconda3/envs/eeg-notebooks/lib/OpenGL' (no such file), '/opt/anaconda3/envs/eeg-notebooks/bin/../lib/OpenGL' (no such file), 'OpenGL' (no such file), '/usr/local/lib/OpenGL' (no such file), '/usr/lib/OpenGL' (no such file), '/Users/priyanshiporwal/eeg-notebooks/OpenGL' (no such file), '/usr/local/lib/OpenGL' (no such file), '/usr/lib/OpenGL' (no such file)", 'OpenGL', None)

Zaduk, 2022-03-15:
Same here, I've been getting the same issue, with the exact same specs as @porwal97. Note: I use an ARM MacBook Air. I've read the main SO article for this error (https://stackoverflow.com/questions/63475461/unable-to-import-opengl-gl-in-python-on-macos) and followed the recommended answer. I've tried changing it to my actual path in my own file system, but nothing.

JohnGriffiths, 2022-03-15:
Hi @porwal97 @Zaduk .

Thanks for the feedback and your interest in `eeg-notebooks`. 

I will try to look into this problem, but unfortunately don't have access to an ARM Mac system for testing. 

All of the OpenGL functionality in `eeg-notebooks` is 'indirect', i.e. it is via `Psychopy`. I don't see the full stack trace from @porwal97 's Import Error, but I would guess it is a `from psychopy import visual, core` line. The best help you can get on that issue will be from the [`Psychopy` forum](https://discourse.psychopy.org/). I recommend post on there the incriminating `Psychopy` line, plus your OS information. 

Definitely keep us posted here on how this goes!

ErikBjare, 2022-03-25:
I had the same issue in CI: https://github.com/NeuroTechX/eeg-notebooks/runs/5690930290

I updated PsychoPy to the latest version in https://github.com/NeuroTechX/eeg-notebooks/pull/167, which solved the issue. Note however that in its place there's a new issue, but there's a known workaround: https://github.com/bambocher/pocketsphinx-python/issues/28

pjkohler, 2022-03-28:
Commenting here to say that I also have issues with OpenGL on a recent (M1) MacBook Air, I assume it is for similar reasons. So if instructions materialize on how to get it working on this type of machine, I would love to see them.

Zaduk, 2022-04-05:
Update: I installed Xcode and it solved the issue for me. I've won, but at what cost...