NeuroTechX/EEG-ExPy #178: InvalidArgumentsError: Unexpected argument of type <class 'type'>

shishankjain, 2022-04-09:
## ℹ Computer information

- Platform OS : Linux (Ubuntu 20.04)
- Python Version:3.7.13
- Brain Interface Used : OpenBCI

## 📝 Provide detailed reproduction steps (if any)

1. …Just installed the eeg notebooks following the first steps mentioned in the installation guide.
2. After executing the test code in the installation guide in the jupyter notebook. Error Present in the screenshot came up. It is also coming up while checking signal.

### ✔️ Expected result
Library should work properly and start checking signal strength for OpenBCI
_What is the expected result of the above steps?_

### ❌ Actual result
InvalidArgumentsError: Unexpected argument of type <class 'type'>
_What is the actual result of the above steps?_

## 📷 Screenshots
[image]
[image]

JohnGriffiths, 2022-04-09:
Hi Shishankjan. 

Thanks for sharing this. 

It's hard to tell from your screenshots what exactly you did ( eg what did you type on the cli) that raised this error. Can you please clarify that ?

shishankjain, 2022-04-09:
For checking signal: eegnb checksiqual -ed cyton

shishankjain, 2022-04-09:
[image]

In the jupyter notebook, I pasted this code. 
Selected kernel, eegnotebooks
and then again same issue popped up.

JohnGriffiths, 2022-04-09:
Ok. Looks like everything is correct on the eeg-notebooks side  but there is an issue with your brainflow installation. 

Can you try:

`pip uninstall brainflow`

`pip install -U brainflow`

`ipython`

```python
from brainflow import BoardShim
```

Does this throw the same error ?

shishankjain, 2022-04-09:
Yes it is also throwing the same error.
[image]

ErikBjare, 2022-04-11:
@shishankjain It it hard to figure out the cause of this without the full logs. Please paste them in full instead of screenshots.

oreHGA, 2023-04-27:
Closing due to inactivity @shishankjain please feel free to re-open if you still require assistance with your brainflow installation

closed 2023-04-27 by oreHGA