NeuroTechX/EEG-ExPy #185: Error after fresh installation

danielemarinazzo, 2022-06-16:
## ℹ Computer information

- Platform OS (e.g Windows, Mac, Linux etc):  Windows
- Python Version: 3.7
- Brain Interface Used (e.g Muse, OpenBCI, Notion etc): Muse2

## 📝 Provide detailed reproduction steps (if any)

1. Did fresh installation, new environment, clone, `pip install -e .`
2. Started quality check

### ✔️ Expected result

Channel quality check as per @JohnGriffiths NTCS videos

### ❌ Actual result

```
  File "C:\Users\daniele\miniconda3\envs\eeg-notebooks\Scripts\eegnb-script.py", line 33, in <module>
    sys.exit(load_entry_point('eeg-notebooks', 'console_scripts', 'eegnb')())
  File "C:\Users\daniele\miniconda3\envs\eeg-notebooks\Scripts\eegnb-script.py", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "C:\Users\daniele\miniconda3\envs\eeg-notebooks\lib\site-packages\importlib_metadata\__init__.py", line 203, in load
    module = import_module(match.group('module'))
  File "C:\Users\daniele\miniconda3\envs\eeg-notebooks\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "c:\users\daniele\dropbox\code\eeg-notebooks\eegnb\cli\__main__.py", line 9, in <module>
    from .introprompt import intro_prompt
  File "c:\users\daniele\dropbox\code\eeg-notebooks\eegnb\cli\introprompt.py", line 6, in <module>
    from eegnb.devices.eeg import EEG
  File "c:\users\daniele\dropbox\code\eeg-notebooks\eegnb\devices\eeg.py", line 17, in <module>
    from brainflow import BoardShim, BoardIds, BrainFlowInputParams
ImportError: cannot import name 'BoardShim' from 'brainflow' (C:\Users\daniele\miniconda3\envs\eeg-notebooks\lib\site-packages\brainflow\__init__.py)
```
This happens whichever command I input, like 
`eegnb checksigqual -ed muse2
`or
`eegnb runexp -ip
`
the muse is streaming ok with bluemuse

## 📷 Screenshots

[image]

ErikBjare, 2022-06-16:
Fixed in https://github.com/NeuroTechX/eeg-notebooks/commit/75e42ef09181a976eac4531a75e9282590a6345e, just cherrypicked onto master.

JohnGriffiths, 2022-06-16:
( FYI this appeared to be due to a small brainflow API change . @danielemarinazzo please try out a re-install with @ErikBjare 's fix. )

danielemarinazzo, 2022-06-16:
thanks, can I just redo `pip install -e . `or shall I undo something?

JohnGriffiths, 2022-06-16:
checked and I also had the error before the fix; and `pip install` now working ok again on my side.

JohnGriffiths, 2022-06-16:
good catch @ErikBjare and @danielemarinazzo , thanks for flagging.

ErikBjare, 2022-06-16:
Fixed in https://github.com/NeuroTechX/eeg-notebooks/commit/f208e086a7302ffec069ed1475faabbf6b95f9d9

There may be follow-up errors, but they deserve separate issues.

danielemarinazzo, 2022-06-16:
works now, thanks!

closed 2022-06-16 by danielemarinazzo