NeuroTechX/EEG-ExPy #191: Error upon end of session

danielemarinazzo, 2022-06-27:
## ℹ Computer information

- Platform OS (e.g Windows, Mac, Linux etc): Windows 10
- Python Version: 3.7
- Brain Interface Used (e.g Muse, OpenBCI, Notion etc): Muse 2

## 📝 Provide detailed reproduction steps (if any)

1. Setup OK
2. Run quality check OK
3. Run N170 experiment

### ✔️ Expected result

ends and save data

### ❌ Actual result

The first time it ended after one second, and actually saved the data, without returning an error

[image]

Then I tried again, this time it completed the full run, but at the end there was an error message, and no data had been saved

[image]

JohnGriffiths, 2022-07-21:
Hi Daniele. 

Thanks for this info. 

We have not seen this error before. 

Are you getting it reliably or is it a one-off?

This is a brainflow error, so Andrey and the brainflow team will be able to give you some more insight into what the cause is, if you can re-post this issue to https://github.com/brainflow-dev/brainflow . Please link to this issue also when you do so we can monitor the discussion.

danielemarinazzo, 2022-07-29:
So it happens every now and then, in my case it was the first time so I thought it was a structural problem. One out of 10 times approx it gives an error, either when selecting the interface (less annoying) or in the middle of the acquisition (more annoying).

I'll close this here and link to the brainflow repo.

thanks!

danielemarinazzo, 2022-07-29:
https://github.com/brainflow-dev/brainflow/issues/529#issuecomment-1199307290