NeuroTechX/EEG-ExPy #192: No Label Column When Running n170 experiment.

Astroneural, 2022-07-07:
When I run the n170 experiment, everything works fine except when I look into the csv file after the experiment is over, I don't see any 'label' or 'stim' or 'marker' column, so I can't run any classifications on it.
To connect to my Muse 2 EEG I run muselsl stream in the terminal (in the same folder that eeg-notebooks is in) and then run the n170 experiment through Jupyter Notebook. I can see all the EEG data with timestamp in the resulting csv, but no label column. 

What should I do to get the data? Is there a way I could do so by modifying the code in n170.py? 

Thank you.

JohnGriffiths, 2022-07-08:
Hi Astro. 

Thanks for getting in touch. 

What OS are you on? Mac/Linux? 

You shouldn't need to start a muselsl stream manually this way. 

If you are running on linux/mac, [the `eeg` class will initiate a stream using muselsl functions for you](https://github.com/NeuroTechX/eeg-notebooks/blob/f208e086a7302ffec069ed1475faabbf6b95f9d9/eegnb/devices/eeg.py#L112) when you start an experiment. 

If you are running on windows then then you could do something like setting up an lsl stream like this with Bluemuse. 

However the shiny new recommended way to initiate a muse stream on windows is via Brainflow, which is done by defining the device type as `muse2016_bfn', 'muse2_bfn' or 'museS_bfn'. 

Does this help?

Best,

j

Astroneural, 2022-07-08:
Hi, thanks for the reply. I installed BlueMuse and initiated the stream through that, however in the file outputted by the Jupyter Notebook I'm still not getting a label column. Also, I am not sure if BlueMuse outputs its own csv file, but I can't find one.

JohnGriffiths, 2022-07-10:
Bluemuse doesn't output its own EEG file, it just creates the data stream. 

Ok let's take a step back. Close bluemuse, don't open it for now, we will try the brainflow option first. Can you please run on the command line 

`eegnb runexp -ip` 

And select n170 expt from the menu, with `muse2_bfn` device option. 

Run duration cam be short e.g. 30s. 

This should produce a file in the `~/.eegnb/.../.../` folder. You can tell which one it is from the datetime stamp in the filename. 

Give this a go and lmk what you observe. When you find the saved file, how many columns does it have ?

Astroneural, 2022-07-10:
[image]
Here's what the file looks like, 7 columns, but the last one has no title. Also, the zeroes go all the way down and do not change. In some of my earlier trials I noticed the same thing, although this extra column doesn't always appear (only sometimes).

JohnGriffiths, 2022-07-10:
Ok. The above appearance is a known bug that occurs with muselsl steaming, and is one of the reasons why we have switched to brai flow as the preferred option. There is a hacky fix for it - see 

https://github.com/NeuroTechX/eeg-notebooks/blob/f208e086a7302ffec069ed1475faabbf6b95f9d9/eegnb/analysis/utils.py#L526

But you shouldn't be getting that problem when using the brainflow streaming option. Can you follow the instructions in my previous message and show the contents of the file produced? Make sure bluemuse is NOT turned on.

JohnGriffiths, 2022-07-11:
Ok, this is good. 

You should find that it is not entirely zeros, there are a relatively small number of 1s and 2s. 

Is that right?

Astroneural, 2022-07-11:
Ah yes! Thank you for pointing them out, I scrolled through too fast before to see them. Thank you so much!

closed 2022-07-14 by Astroneural