NeuroTechX/EEG-ExPy #196: Test Experimental Class Structure using EEG

Parvfect, 2022-07-21:


closed 2023-04-27 by oreHGA

LIN-DN, 2023-07-19:
Have you tested the updated code now? Regarding the visualN170 experiment, the documentation does not appear to have updated the case code. n170.present(duration=record_duration, eeg=eeg_device, save_fn=save_fn) module 'eegnb.experiments.visual_n170.n170' has no attribute 'present'.

JohnGriffiths, 2023-07-19:
Thanks for noting this @LIN-DN . The `.present` method is now replaced with then`run` method. However it sees the example notebook in the `master` branch is still showing the old syntax. In the `develop` branch it is correct. 

@oreHGA @tmorshed @Parvfect I think seeing as this is a key documentation file it should be fixed with a direct commit to `master` rather than waiting for the next merge of `develop` to master. What do you think? Pls check and confirm I got this correct.