NeuroTechX/EEG-ExPy #199: Problems in data streaming Muse2 for N170 and P300

Parvfect, 2022-07-26:
### Discussed in https://github.com/NeuroTechX/eeg-notebooks/discussions/198

<div type='discussions-op-text'>

<sup>Originally posted by **Parvfect** July 26, 2022</sup>
I am trying to run the N170 experiment, and it seemed to be working perfectly before and I have some good data from then. I am not sure what I changed, but the instruction screen seems to come up twice, the second time after running the main experiment for a few seconds.  About two images get displayed when the same screen comes up for the second time. The code used is the following 

`from eegnb.devices.eeg import EEG
from eegnb.experiments.visual_n170 import n170

board_name = 'muse2'

record_duration = 120
eeg_device = EEG(device=board_name)
n170.present(duration=record_duration, eeg=eeg_device)`

And the runtime error is the following, 

[image]
[image]

[image]

I am not very clear why this is happening and would really apprectiate some help. I was testing the ExperimentSubClass2 branch when this happened and switched to Master and got the same issue. 
</div>

JohnGriffiths, 2022-07-26:
Can you repeat and show screen outputs for device = `muse2_bfn` ?

Parvfect, 2022-07-27:
I get a key-error when I run with muse2_bfn as the device. Am I doing something wrong?

JohnGriffiths, 2022-07-27:
Yes something not right if you're using the above script and getting a key error.

We need to get back to step 1. 

Please try:

- fresh clone of eeg-notebooks (rename as '_temp' or '_v2' or whatev)
- go in to folder, create new environment, activate environment, `pip install -e .`
- Turn on muse 
- On command line `eegnb checksigqual -ed muse2_bfn`
- If not working: repeat a few times; make sure bluetooth is on
- If you get errors from this, please share screenshot here.

Parvfect, 2022-07-28:
So it works with muse2 (the signal quality check), but if I do muse2_bfn, I get this
[image]

JohnGriffiths, 2022-07-28:
Do you mean it works with muse2 when bluemuse is running? 

Try one more time: close conda terminal, close bluemuse, open new conda terminal, run _bfn signal check. 

(Brainflow can get confused if there's been other recent communication with the device by currently running programs)

Parvfect, 2022-07-28:
Yeah so it works with bluemuse and muse2. On closing bluemuse and independently running brainflow via the _bfn run check, I get this error as before
[image]

Parvfect, 2022-08-03:
https://github.com/brainflow-dev/brainflow/issues/531

Issue opened in Brainflow

Unable to prepare the streaming session after finding Muse2 using direct calls to Brainflow.

Code: 

```python
from brainflow.board_shim import BoardShim, BoardIds, BrainFlowInputParams

params = BrainFlowInputParams()
board_id = BoardIds.MUSE_2_BOARD

board = BoardShim(board_id, params)
board.enable_dev_board_logger()

board.prepare_session()
```

Screenshot of Error
[image]

Parvfect, 2022-08-03:
Original problem with respect to data streaming has been solved by using BlueMuse backend. The problem was that my file had an open call to streaming which was fixed by adding the following, 

```python

if __name__ == '__main__':
    # Do the Experiment

```

However, setting up a streaming session using Brainflow is still unresolved and listed above.

oreHGA, 2023-04-27:
Summary:
- Suggestion from linked brainflow thread https://github.com/brainflow-dev/brainflow/issues/531 was to update to 3.10