NeuroTechX/EEG-ExPy #200: Installing eeg notebooks on Mac M1

oreHGA, 2022-07-28:
Hey folks, wanted to use this thread to keep track of the changes to support installation of eeg-notebooks on Mac M1.

(I initially tried using poetry for this but it added another layer of complexity w the errors so I decided to get it working with manual pip first)

I was able to tweak the requirements.txt enough to get installation to complete, but I get this error when trying to run `eegnb runexp -ip` for an experiment

```bash
  File "/Users/oreogundipe/miniconda3/envs/eeg-notebooks/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 843, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/Users/oreogundipe/lab/eeg-notebooks/eegnb/cli/__main__.py", line 9, in <module>
    from .introprompt import intro_prompt
  File "/Users/oreogundipe/lab/eeg-notebooks/eegnb/cli/introprompt.py", line 7, in <module>
    from .utils import run_experiment, get_exp_desc, experiments
  File "/Users/oreogundipe/lab/eeg-notebooks/eegnb/cli/utils.py", line 15, in <module>
    from eegnb.experiments.auditory_oddball import aob, diaconescu
  File "/Users/oreogundipe/lab/eeg-notebooks/eegnb/experiments/auditory_oddball/aob.py", line 3, in <module>
    from psychopy import visual, core, event, sound
  File "/Users/oreogundipe/miniconda3/envs/eeg-notebooks/lib/python3.8/site-packages/psychopy/sound/__init__.py", line 99, in <module>
    raise exceptions.DependencyError(
psychopy.exceptions.DependencyError: No sound libs could be loaded. Tried: ['PTB']
Check whether the necessary sound libs are installed
0.0019 	WARNING 	PTB audio lib was requested but not loaded: psychtoolbox audio failed to import
```

To get things running on my mac, I'm eegnb install working I'm using latest beta release for psychtoolbox 3.0.18.10 (https://github.com/Psychtoolbox-3/Psychtoolbox-3/releases)

All roads point towards psychtoolbox/ psychopy as area to try tweaking
- I've tried using the latest version of psychopy but there's a bug with the installation (https://github.com/psychopy/psychopy/issues/5016). I'm exploring trying to clone a local build to get it working

@JohnGriffiths do you have any pointers on how to get `soundlibs` installed? I'll keep this thread updated if anything gives

JohnGriffiths, 2022-07-28:
Is `psychtoolbox` installed? Looks like it isn't finding it ?

More generally, if this failure is going to occur it should happen when you are trying to run the auditory experiment, not at the experiment selection stage.

pellet, 2025-05-01:
We now have instructions on how to set up and install an environment on MacOS ARM (M1) https://neurotechx.github.io/EEG-ExPy/getting_started/installation.html#installing-the-python-library

oreHGA, 2025-05-03:
Thanks @pellet!

closed 2025-05-03 by oreHGA