NeuroTechX/EEG-ExPy #205: N170 Load and Visualize Data

Liam-Noah, 2022-09-12:
## 📝 Provide a description of requested docs changes

https://neurotechx.github.io/eeg-notebooks/auto_examples/visual_n170/01r__n170_viz.html
Hello,
Could you explain how eegnb_data_path and n170_data_path variable are and how to use them ?
I set  n170_data_path to be 
n170_data_path = "C:/Users/.../.eegnb/data/visual_n170/local/muse2/subject0000/session000/recording_2022-09-12-15.40.21.csv"
but apparently I have to change eegnb_data_path aswell ?

_What is the purpose and what should be changed?_

JohnGriffiths, 2022-09-14:
Hi Vladimir. Can you screenshot the first 30 lines or so of the csv file ? Is it missing the final column in the first few rows?

Liam-Noah, 2022-09-15:
> Hi Vladimir. Can you screenshot the first 30 lines or so of the csv file ? Is it missing the final column in the first few rows?

Yes, that's exactly what's happening here 

[image]

JohnGriffiths, 2022-09-15:
Ok. This is a known bug that occurs sometimes with muselsl streaming that has to do with asynchronous recording process initialization, and that we haven't managed to get a robust fix to yet unfortunately.

Sorry this has been a problem.

The data you have should be usable, the file just needs repairing. There is a function in the library for this:

https://github.com/NeuroTechX/eeg-notebooks/blob/a22501228677df0f968c49a0a699207380b1bd50/eegnb/analysis/utils.py#L530

usage ftom a python interpreter:

```python
from eegnb analysis.utils import fix_musemissinglines

fpath = " " # path to the .csv file
fix_musemissinglines(fpath)
```

Which will create a new file with '_fml.csv' as suffix in the same folder.

Moving forward, you have two options:

1. Continue as you have been, and applying this fix, to recorded files

2. Switch streaming options 

I highly recommend 2. The brainflow streaming options do not have this issue.  You would give device option `muse2_bfn` ( bfn stands for BrainFlow with Native bluetooth hardware of your system, as opposed to brainflow with a BLED dongle which is '_bfb' ). The only reason why this isn't the only option listed is there are some unsolved brainflow issues with the newest  windows 11 distributions. But you should try and see if this works on your comp.

What OS are you on?

Liam-Noah, 2022-09-15:
Hi again, 
I'm on windows 10. I'll give brainflow a try since there are less bugs. Thanks for the tips ! 

Do you real time analysis with jupyter notebook aswell? To have a neurofeedback window with live parameters, to see how well you do on getting focused or calm .

oreHGA, 2022-12-15:
Hi @Liam-Noah , eeg-notebooks doesn't currently support realtime analysis but here are a couple links that could help
- EEGEdu - https://eegedu.com/
- Timeflux - https://timeflux.io/

Some headsets like [Neurosity](https://neurosity.co) has an api that can give you predictions every second of focus/calm

closed 2023-04-27 by oreHGA