NeuroTechX/EEG-ExPy #212: Error with installing durkp

WeiShaoD, 2022-11-14:
## ℹ Computer information

Hello, everyone, I met a problem when I tried to install with pip3 install dukpy package (I need this module to install psychopy )

- MacOS Ventura 13.0 M1, 
- Python Python Version: Python 3.9.12

If there any way to fix this? Thanks

(base) weishao@wei 2021_experiment % python3 --version
Python 3.9.12
(base) weishao@wei 2021_experiment % pip3 install dukpy       
Collecting dukpy
  Using cached dukpy-0.2.3.tar.gz (1.9 MB)
Building wheels for collected packages: dukpy
  Building wheel for dukpy (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /Users/weishao/Documents/y/fslpython/bin/python -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/m8/z0y3_n3x0bldp2fd5xpz9jjm0000gn/T/pip-install-24gxh4yq/dukpy_bd0530a693ec47cbb5d46fc7c73f5f21/setup.py'"'"'; __file__='"'"'/private/var/folders/m8/z0y3_n3x0bldp2fd5xpz9jjm0000gn/T/pip-install-24gxh4yq/dukpy_bd0530a693ec47cbb5d46fc7c73f5f21/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/var/folders/m8/z0y3_n3x0bldp2fd5xpz9jjm0000gn/T/pip-wheel-hkhuiz27
       cwd: /private/var/folders/m8/z0y3_n3x0bldp2fd5xpz9jjm0000gn/T/pip-install-24gxh4yq/dukpy_bd0530a693ec47cbb5d46fc7c73f5f21/
  Complete output (145 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.macosx-10.9-x86_64-3.9
  creating build/lib.macosx-10.9-x86_64-3.9/dukpy
  copying dukpy/run.py -> build/lib.macosx-10.9-x86_64-3.9/dukpy
  copying dukpy/__init__.py -> build/lib.macosx-10.9-x86_64-3.9/dukpy
  copying dukpy/lessc.py -> build/lib.macosx-10.9-x86_64-3.9/dukpy
  copying dukpy/evaljs.py -> build/lib.macosx-10.9-x86_64-3.9/dukpy
  copying dukpy/babel.py -> build/lib.macosx-10.9-x86_64-3.9/dukpy
  copying dukpy/install.py -> build/lib.macosx-10.9-x86_64-3.9/dukpy
  copying dukpy/coffee.py -> build/lib.macosx-10.9-x86_64-3.9/dukpy
  copying dukpy/tsc.py -> build/lib.macosx-10.9-x86_64-3.9/dukpy
  copying dukpy/module_loader.py -> build/lib.macosx-10.9-x86_64-3.9/dukpy
  copying dukpy/nodelike.py -> build/lib.macosx-10.9-x86_64-3.9/dukpy
  creating build/lib.macosx-10.9-x86_64-3.9/dukpy/webassets
  copying dukpy/webassets/jsxfilter.py -> build/lib.macosx-10.9-x86_64-3.9/dukpy/webassets
  copying dukpy/webassets/__init__.py -> build/lib.macosx-10.9-x86_64-3.9/dukpy/webassets
  copying dukpy/webassets/typescriptfilter.py -> build/lib.macosx-10.9-x86_64-3.9/dukpy/webassets
  copying dukpy/webassets/lessfilter.py -> build/lib.macosx-10.9-x86_64-3.9/dukpy/webassets
  copying dukpy/webassets/babelfilter.py -> build/lib.macosx-10.9-x86_64-3.9/dukpy/webassets
  creating build/lib.macosx-10.9-x86_64-3.9/dukpy/jscore
  copying dukpy/jscore/punycode.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jscore
  copying dukpy/jscore/querystring.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jscore
  copying dukpy/jscore/path.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jscore
  copying dukpy/jscore/url.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jscore
  copying dukpy/jscore/fs.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jscore
  creating build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules
  copying dukpy/jsmodules/typescriptServices.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules
  copying dukpy/jsmodules/semver.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules
  copying dukpy/jsmodules/babel-6.26.0.min.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules
  copying dukpy/jsmodules/coffeescript.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules
  creating build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/react
  copying dukpy/jsmodules/react/react.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/react
  copying dukpy/jsmodules/react/react-dom-server.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/react
  creating build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less
  creating build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less
  copying dukpy/jsmodules/less/less/logger.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less
  copying dukpy/jsmodules/less/less/parse-tree.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less
  copying dukpy/jsmodules/less/less/import-manager.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less
  copying dukpy/jsmodules/less/less/render.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less
  copying dukpy/jsmodules/less/less/index.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less
  copying dukpy/jsmodules/less/less/transform-tree.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less
  copying dukpy/jsmodules/less/less/source-map-output.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less
  copying dukpy/jsmodules/less/less/contexts.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less
  copying dukpy/jsmodules/less/less/parse.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less
  copying dukpy/jsmodules/less/less/plugin-manager.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less
  copying dukpy/jsmodules/less/less/utils.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less
  copying dukpy/jsmodules/less/less/source-map-builder.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less
[... 229 lines cut ...]
    copying dukpy/jsmodules/less/less/functions/types.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less/functions
    copying dukpy/jsmodules/less/less/functions/function-registry.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less/functions
    copying dukpy/jsmodules/less/less/functions/data-uri.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less/functions
    copying dukpy/jsmodules/less/less/functions/index.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less/functions
    copying dukpy/jsmodules/less/less/functions/color.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less/functions
    copying dukpy/jsmodules/less/less/functions/string.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less/functions
    copying dukpy/jsmodules/less/less/functions/math.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less/functions
    copying dukpy/jsmodules/less/less/functions/color-blending.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less/functions
    copying dukpy/jsmodules/less/less/functions/svg.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less/functions
    copying dukpy/jsmodules/less/less/functions/default.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less/functions
    copying dukpy/jsmodules/less/less/functions/function-caller.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less/functions
    creating build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less/data
    copying dukpy/jsmodules/less/less/data/index.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less/data
    copying dukpy/jsmodules/less/less/data/colors.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less/data
    copying dukpy/jsmodules/less/less/data/unit-conversions.js -> build/lib.macosx-10.9-x86_64-3.9/dukpy/jsmodules/less/less/data
    running build_ext
    building 'dukpy._dukpy' extension
    creating build/temp.macosx-10.9-x86_64-3.9
    creating build/temp.macosx-10.9-x86_64-3.9/src
    creating build/temp.macosx-10.9-x86_64-3.9/src/duktape
    clang -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /Users/weishao/Documents/y/fslpython/include -arch x86_64 -I/Users/weishao/Documents/y/fslpython/include -fPIC -O2 -isystem /Users/weishao/Documents/y/fslpython/include -arch x86_64 -DDUK_OPT_DEEP_C_STACK=1 -DDUK_OPT_NONSTD_REGEXP_DOLLAR_ESCAPE=1 -DDUK_OPT_OCTAL_SUPPORT=1 -I./src/duktape -I/Users/weishao/Documents/y/fslpython/include/python3.9 -c src/_support.c -o build/temp.macosx-10.9-x86_64-3.9/src/_support.o
    xcrun: error: invalid active developer path (/Library/Developer/CommandLineTools), missing xcrun at: /Library/Developer/CommandLineTools/usr/bin/xcrun
    error: command '/usr/bin/clang' failed with exit code 1
    ----------------------------------------
[... cut ...] , open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/m8/z0y3_n3x0bldp2fd5xpz9jjm0000gn/T/pip-record-cznpetwo/install-record.txt --single-version-externally-managed --compile --install-headers /Users/weishao/Documents/y/fslpython/include/python3.9/dukpy Check the logs for full command output.

ErikBjare, 2022-11-24:
Potential duplicate of: https://github.com/NeuroTechX/eeg-notebooks/issues/96

JohnGriffiths, 2022-12-15:
Hey @WeiShaoD - did you try @ErikBjare 's suggestion of installing XCode developer tools?

tmorshed, 2023-04-06:
Closing due to no-response. Feel free to reopen if the issue is still going on.