NeuroTechX/EEG-ExPy #213: Can't run experiments from CLI interface

Hororohoruru, 2022-11-18:
## ℹ Computer information

- Platform OS (e.g Windows, Mac, Linux etc):  Mac
- Python Version: 3.8
- Brain Interface Used (e.g Muse, OpenBCI, Notion etc): Muse (2016)

## 📝 Provide detailed reproduction steps (if any)

1. Initialize Muse stream using command ```muselsl stream```
2. Open a separate terminal and launch the CLI with ```eegnb runexp -ip```
3. Introduce the following input: 
- Device: Muse (2016)
- Experiment: N170 (option 0)
- Duration: 60
- Participant ID: 1
- Run number: 1

### ✔️ Expected result

The experiment runs correctly

### ❌ Actual result

The Muse device is found, and the experiment window opens correctly. However, when the spacebar is pressed to start the experiment, nothing happens for some time and then the window halts and exits. Here is the traceback from the terminal:

```
Searching for Muses, this may take up to 10 seconds...
No Muses found.
Traceback (most recent call last):
 File "/Users/f.dehais/opt/anaconda3/envs/eeg-notebooks/bin/eegnb", line 8, in <module>
  sys.exit(main())
 File "/Users/f.dehais/opt/anaconda3/envs/eeg-notebooks/lib/python3.7/site-packages/click/core.py", line 1130, in __call__
  return self.main(*args, **kwargs)
 File "/Users/f.dehais/opt/anaconda3/envs/eeg-notebooks/lib/python3.7/site-packages/click/core.py", line 1055, in main
  rv = self.invoke(ctx)
 File "/Users/f.dehais/opt/anaconda3/envs/eeg-notebooks/lib/python3.7/site-packages/click/core.py", line 1657, in invoke
  return _process_result(sub_ctx.command.invoke(sub_ctx))
 File "/Users/f.dehais/opt/anaconda3/envs/eeg-notebooks/lib/python3.7/site-packages/click/core.py", line 1404, in invoke
  return ctx.invoke(self.callback, **ctx.params)
 File "/Users/f.dehais/opt/anaconda3/envs/eeg-notebooks/lib/python3.7/site-packages/click/core.py", line 760, in invoke
  return __callback(*args, **kwargs)
 File "/Users/f.dehais/eeg-notebooks/eegnb/cli/__main__.py", line 80, in runexp
  run_experiment(experiment, eeg, recdur, outfname)
 File "/Users/f.dehais/eeg-notebooks/eegnb/cli/utils.py", line 45, in run_experiment
  module.present(duration=record_duration, eeg=eeg_device, save_fn=save_fn) # type: ignore
 File "/Users/f.dehais/eeg-notebooks/eegnb/experiments/visual_n170/n170.py", line 56, in present
  eeg.start(save_fn, duration=record_duration + 5)
 File "/Users/f.dehais/eeg-notebooks/eegnb/devices/eeg.py", line 410, in start
  self._start_muse(duration)
 File "/Users/f.dehais/eeg-notebooks/eegnb/devices/eeg.py", line 120, in _start_muse
  target=stream, args=(self.muses[0]["address"],)
IndexError: list index out of range
2022-11-18 14:40:34.081 python[32064:8357601] IMKClient Stall detected, *please Report* your user scenario attaching a spindump (or sysdiagnose) that captures the problem - (imkxpc_bundleIdentifierWithReply:) block performed very slowly (11.38 secs).
2022-11-18 14:40:34.081 python[32064:8357601] IMKClient Stall detected, *please Report* your user scenario attaching a spindump (or sysdiagnose) that captures the problem - (imkxpc_bundleIdentifierWithReply:) block performed very slowly (11.37 secs).
```

In fact, after checking the code on eegnb/devices/eeg.py, I saw that self.muses is being assigned using the ```list_muses()``` method. However, if the device is already streaming, the function cannot find it. I tried to start a stream and then run ```muselsl list``` from a separate terminal. I assume the error is thrown after the code fails to find any devices after the 10 second timeout and then the list is empty at line 120. I found a similar problem in https://github.com/NeuroTechX/eeg-notebooks/issues/132#issuecomment-1076820751, but we would like to use the CLI, so the solution found by the author of that issues is not possible for our purposes.

It is possible that I am missing something on how the Muse is supposed to work (not an expert with the device). If I understand correctly, the first steps on the CLI need a stream to be running, but then the experiment tries to connect to a device that is not running, and it does not find it. I would really appreciate your input on the problem.

Thank you very much for the library, we intend to use it for educational purposes and would love to fix this issue in order to share it with our students.

oreHGA, 2022-12-15:
Hey @Hororohoruru , can you try just running `eegnb runexp -ip` without starting a muselsl stream yourself?

The library [starts a stream under the hood](https://github.com/NeuroTechX/eeg-notebooks/blob/a22501228677df0f968c49a0a699207380b1bd50/eegnb/devices/eeg.py#L112) for you when the command is run.

Our current guess is that having two of these running is leading to a conflict since the port is already in use.

tmorshed, 2023-04-06:
@Hororohoruru Hello there! Any updates on this issue? Is it resolved or is it still going?

Hororohoruru, 2023-04-13:
Hello! Sorry for the late reply. We were on the clock to prepare some teaching materials, so we never got to test the suggested solution, sorry.

oreHGA, 2023-04-27:
No worries @Hororohoruru closing this for now then. Please feel free to re-open if you're still running into issues

closed 2023-04-27 by oreHGA