NeuroTechX/EEG-ExPy #214: pyglet version incompatibility mac

Silkedesmet, 2022-11-30:
## ℹ Computer information

- Platform OS (e.g Windows, Mac, Linux etc):  Mac
- Python Version: 3.7
- Brain Interface Used (e.g Muse, OpenBCI, Notion etc): Muse2

## 📝 Provide detailed reproduction steps (if any)

1. created environment on miniconda with python 3.7 
2. cloned repo
3. installed
4. run `eegnb runexp -ip`

### ✔️ Expected result

_What is the expected result of the above steps?_

cli opens

### ❌ Actual result

`pyglet 2.0.0 requires Python 3.8 or newer`

[image]

If on the other hand I try to create an environment with python 3.8 I get some package conflict errors

[image]

sotofernando, 2023-02-15:
Hi, same issue for me: 
`Exception: pyglet 2.0.4 requires Python 3.8 or newer.`

any ideas? thanks

sotofernando, 2023-03-07:
This is what I did (don't know if every step is correct/necessary)

I stick with python 3.7, and install legacy pyglet (don't require python 3.8+, but 3.6+)
```
pip install pyglet==1.5.27
pip install PyOpenGL PyOpenGL_accelerate
```
As pointed out in https://github.com/PixarAnimationStudios/USD/issues/1372#issuecomment-809929362
I edit PyOpenGL file `OpenGL/platform/ctypesloader.py`

And then, run interactive experiment in terminal (I select synthetic board, n170, and "no" for "Run signal quality check?")

`eegnb runexp -ip
`

It shows faces/houses images and saves data from board to csv.

danielemarinazzo, 2023-03-07:
> As pointed out in PixarAnimationStudios/USD#1372 (comment)

Great to hear.
Maybe with the new 3.8 version the problem will be fixed anyway?

sotofernando, 2023-03-08:
Maybe, I didn't look for working around package conflicts when installing conda environment with python 3.8