NeuroTechX/EEG-ExPy #216: Implement version + release schedule

JohnGriffiths, 2023-01-05:
We are going to merge the develop branch to master, which contains two main new components: the refactored experiment class and the analysis report function. 

As part of doing this, we will implement a more structured release schedule process, which will allow us to more easily keep on top of changes and comfortably execute releases / merge updates to master that *might* possibly cause backwards compatibility issues. 

Overall tasks:

- [ ] Confirm we are ready to move develop branch contents into master  
   - [ ] Check the code runs (again)
   - [ ] Check for any API changes from previous version, and either adjust if possible or document
   - [ ] Write docs on new features (analysis report; maybe others)
   - [ ] Update docs on anything major that's missing
   - [ ] Check docs in general; esp. installation
- [ ] Write doc describing this version plan  
- [ ] Create new branch from current master, called 0.2 (current version)  
- [ ] Create new branch from develop, called 0.3 (new version)  
   - [ ] Check requirements are up to date
   - [ ] Make changelog file
- [ ] Merge 0.3 to master  
- [ ] Add pypi installation for 0.3 onwards  
- [ ] Add multi-version doc build configuration (c.f. github.com/mne-tools.github.io)

tmorshed, 2023-01-19:
**Problem:**
Regarding the first action item, we see that `Develop` fails building the docs. 

**Investigation:**
Started a new codespaces instance and checked it all. Still failed. The only thing that needed to be changed was the linux version. Seems like GH moved to Ubuntu 22.04 for actions since Nov. 2022 (which is around the time that this issue started). 

**Solution**:
Updating the version number of the `wxpython` wheel to 22.04. i.e.:
```
pip install -U -f https://extras.wxpython.org/wxPython4/extras/linux/gtk3/ubuntu-18.04 wxPython
```
must become 
```
pip install -U -f https://extras.wxpython.org/wxPython4/extras/linux/gtk3/ubuntu-22.04 wxPython
```
Will try the update and see if it fixes the issue.

Parvfect, 2023-01-19:
*Problem with respect to Docs*

Docs tutorials for Running Experiments deal with importing individual experiements and then running them as per, 
[image]

However, with the Experiment Class Refactor #184, with classes and objects and such, the call changes slightly and will not work with the current code. Going to address this in a new pull request to master once the merge is done to fix the problem.